keyboardio/Chrysalis #36: Persist data between pages

algernon, 2018-12-02:
Right now, state is local to the pages. When one switches from the info page to the layout editor, it will query the keyboard. Switching back to info, then back to layout editor again, and the query will be performed again. This is slow, and inefficient. Since we're locking the serial port for ourselves, it is safe to persist this data somewhere globally.

We do need to provide a way to purge this cached info. We won't pre-load all of it either, the initial load will still happen on the first page visit. But the next visit should not do another query.

algernon, 2018-12-02:
I suppose this is where React Contexts come into play? That, or Redux, I guess.

algernon, 2018-12-02:
[This](https://discuss.reactjs.org/t/vanilla-react-without-redux-or-flux/3733) seems a lot easier than Redux. But Redux has the huge advantage of having awesome developer tools. Do we need those at this point, I wonder?

algernon, 2018-12-04:
Well, `constate` doesn't appear to be working as I had hoped, so I guess I'll just go with `Redux`.

algernon, 2018-12-04:
Got Redux mostly set up, but things fail miserably when trying to use Redux DevTools. Tired of fighting with this, so going to postpone this for a while.

algernon, 2018-12-13:
Taking this off my queue, because with recent changes, Chrysalis is fast enough so that pulling data from the keyboard on each page change isn't a big deal anymore.

algernon, 2018-12-22:
...and closing it, because there's no real benefit of this right now.

algernon, 2019-01-25:
Reopening this, because some of the things we want to do could greatly benefit from a more centralized state. It would also simplify a lot of the code, too.

algernon, 2019-01-25:
On one hand, I like Redux for its devtools. On the other hand, I had trouble setting it up, and Redux needs a lot of boilerplate.

Someone (@TreTuna I think?) mentioned [MobX](https://github.com/mobxjs/mobx), and I started looking at it... it looks interesting. Simple, not much boilerplate.

TreTuna, 2019-01-26:
I would first look at just using React Context, then perhaps MobX, then Redux... I feel like that's the order of complexity you add to your code with those tools.

I prefer using built in methodology (Context) until you really run into an issue that's difficult with it, then really evaluating between MobX and Redux. Once the new React Hooks API is fully stable (or we could even use the alpha now, MANY people are in production) it makes things even easier.

TreTuna, 2019-01-26:
I'm going to be digging into the flow of the code a lot more this weekend and will have a better idea of a recommendation for how to structure persistent state at that point.

TreTuna, 2019-03-07:
I want to start looking at this again, especially since we want it for the 0.7.0 milestone. The big thing that I need to understand more is the Focus package. We call `new Focus()` in 15 different places. Could this be something that is simplified by calling once, keeping this focus in global state, and passing that around instead and just updating that single focus?

TreTuna, 2019-03-07:
I think using React Context with Hooks will be a good way to get going on this. I was also thinking of having two separate Contexts to work with here. An `AppContext` and `DeviceContext`rather than a single bucket for both.

algernon, 2019-03-09:
Regarding Focus: yeah, if we kept it in global state and passed that around, that would make a few things simpler. However, since it is a singleton anyway, there's not much harm in calling `new Focus()` many times, either.

>  An `AppContext` and `DeviceContextrather` than a single bucket for both.

I'm liking this idea.

algernon, 2021-12-13:
I've just had a very simple idea how we could go about doing this, without complicating our codebase with React Contexts and whatnot. We could just cache this state in `sessionStorage`, perhaps even at a `Focus` level. We do not need to have one screen re-render when another screen changes things: that doesn't happen ever in our current architecture.

Using `sessionStorage` would be _simple_, and require very few changes. I'm going to explore that idea soon, probably after finishing #730.

obra, 2021-12-13:
(I expect that there's application state that doesn't make sense to push
into Focus.)

At a certain level, I could see the same solution getting used for
application state and for modeling keyboard state. I'd love it if this gets
built in a way that lets Chrysalis run without a keyboard attached.

obra, 2021-12-13:
> There's not much application state we'd need to share between screens, but
> yeah, if any, that should not be pushed to Focus. Nothing's stopping us for
> using sessionStorage for both, though.
>
> As for running without a keyboard: that's more to do with how api/focus
> is built at the moment than with state. That part of the code relies on
> serial *heavily*, and has little to not abstractions. State is by far the
> easier part of the whole, as far as virtual keyboard support is concerned.

Well, I guess my thought is that api/focus should end up only getting
called through the state management abstraction