keyboardio/Chrysalis #78: Figure out a better way to present the features in the SpeedDial FAB

algernon, 2018-12-19:
Thanks to @davidaue for digging up the Material Design [recommendation](https://material.io/design/components/buttons-floating-action-button.html#usage), namely: "Only use a FAB if it is the most suitable way to present a screen’s primary action."

The stuff in the speed dial does not meet that requirement. As such, we should figure out a better way to present those things. Perhaps an action bar at the bottom? A thin one? That could still go on every page globally.

davidaue, 2018-12-19:
Maybe and overflow menu?
[Top bar Anatomy](https://material.io/design/components/app-bars-top.html#anatomy)

algernon, 2018-12-19:
Yeah, that could work, but not all pages have an appbar right now, which is why I was thinking about a bottom action area instead.

Something like an [Expanding Bottom Sheet](https://material.io/design/components/sheets-bottom.html#expanding-bottom-sheet)? That sounds like something appropriate, and I can fit it onto all pages with ease, and it wouldn't look too out of place.

algernon, 2018-12-20:
I think I'll be going with a menu in the lower right corner, that expands upwards.

PR #90 (2018-12-21): Turn the FAB SpeedDial into a persistent menu
This turns the FAB SpeedDial into a menu that's always on the lower right corner, and functions like a proper menu.

This is done in a three step process:

- First we extract the Tools speed dial into its own component.
- Then we turn it from a FAB into a Menu.
- Then we make it not toggle DevTools, because that's counter-intuitive.

The end result looks like this:

[image]
[image]

Fixes #78.
commit d9c9abc3 (2018-12-21): AppTools: Transition from a FAB to a Menu

Fixes #78.
closed 2018-12-21 by obra