keyboardio/Chrysalis #80: Logical layer abstraction

davidaue, 2018-12-19:
The way layer storage works internally could potentially be confusing to users if the UI presents things as a 1:1 mapping to the internal implementation. Maybe there is a way to elegantly abstract what is happening with layers so novice users get a more intuitive experience. 

* All layers could be referred to by a label instead of a true layer number so that moving them seems not to change anything.
* If a PROGMEM layer is edited it would be copied to an unused EEPROM layer and all references to the layer would be automatically adjusted.
* An "Add Layer" action would allocate an unused EEPROM layer as a new logical layer.

This obviously imposes tighter limitations of what is possible for an advanced user. Perhaps there could be an advanced mode where true layer numbers are shown and no automatic copying happens.

Moving PROGMEM layers into EEPROM to make them editable would break any references to that layer in the firmware that are not dynamically updateable. Could Kaleidoscope could be enhanced to have layer indirection as a base concept so layers being moved would affect the entire system transparently.

algernon, 2018-12-19:
I'm liking these ideas, they make sense, and present a workflow that sounds convenient. We already have a - currently mostly unused - Settings page, we could allow one to turn on "Advanced mode" there (perhaps with a more... descriptive name), and default to the more-hand-holding experience.

algernon, 2019-02-05:
I think that with #269, which implemented the scenario in #228, we have provide a much more straightforward experience, with many of the ideas outlined here integrated into a similar-in-spirit, but a little different goal.

As such, I'm closing this issue. Thank you for the ideas! :)

Layer naming is not possible yet, we have #3 open for that.

closed 2019-02-05 by algernon