keyboardio/Chrysalis #84: When we're showing the "programming the keyboard" spinning progress indicator, we should disable the top-of-page blue progress indicator

obra, 2018-12-20:


algernon, 2018-12-20:
The top-of-page blue progress indicator is - as far as I see - unrelated. That one's shown when we're trying to query the keyboard with a `version` focus command. I guess that never finishes?

obra, 2018-12-20:
Ah. yeah. that must be it. Maybe it needs a timeout?

algernon, 2018-12-20:
It... should time out, eventually, after about 15 seconds. (I'll lower the timeout soon)

Can you verify that it stops the top-blue animation after about 15 seconds? And if it doesn't, is there anything on the dev console?

obra, 2018-12-20:
It was there for multiple minutes. Unfortunately, after flashing the
firmware, I got to a white screen and couldn't enable the dev console. I'll
try to repro later today.

algernon, 2018-12-22:
I just tried to reproduce this, with #95 applied, using `Model01-Firmware`'s master branch (thus, no `version` command), and the blue progress bar wasn't shown during flash. I couldn't even see if the bar even appeared...

algernon, 2018-12-22:
Oooh, I managed to reproduce it. I had another app steal serial from Chrysalis, and now I have LinearProgress stuck on the top permanently.

PR #99 (2018-12-23): KeyboardInfo: Handle errors from focus.command("version")
When querying the version and Focus throws an error, catch that, so we can disable the progress bar.

Fixes #84.
commit ee0a990a (2018-12-23): KeyboardInfo: Handle errors from focus.command("version")

When querying the version and Focus throws an error, catch that, so we can
disable the progress bar.

Fixes #84.
closed 2018-12-23 by obra