keyboardio/Chrysalis #85: The key type dropdown should show a drop-down indicator like a triangle

obra, 2018-12-20:
[image]

algernon, 2018-12-20:
Hm.. this will be a tougher one, because the UI library we use doesn't have a widget like that out of the box. [Material design](https://material.io/design/components/menus.html#exposed-dropdown-menu) mentions dropdowns as a special case of Menu, however.

It shouldn't be *too* hard to add that little dropdown indicator, but it will take a bit longer, 'cos it's not just using another widget.

obra, 2018-12-20:
even adding a ... after the letters for now might help. and coloring it the theme color maybe?

algernon, 2018-12-20:
That should be much easier

algernon, 2018-12-20:
Turns out adding a triangle there was easier than coloring. :)

PR #88 (2018-12-20): KeySelector: Add a dropdown indicator to the key type menu
Fixes #85.
commit e3d3b9d6 (2018-12-20): KeySelector: Add a dropdown indicator to the key type menu

Fixes #85.
closed 2018-12-20 by obra