keyboardio/Chrysalis #112: White Window on Launch

gavinhoward, 2018-12-31:
On Ubuntu 18.04, x86_64, using the 0.2.0 AppImage, I just get a white window when launching the application. The title bar is the only thing that is not white.

I am running on a machine with an NVidia 1080TI, using the NVidia binary driver, if that helps.

algernon, 2018-12-31:
:(

I guess I should make it pop up the devtools when there's an exception during startup. Are you up for trying to run it from source, with a tiny modification? (we'd make it pop  up the developer tools by default, so that you can look at the console, and we can figure out why it fails to start)

algernon, 2018-12-31:
If you have node & yarn installed, it's pretty easy. If you don't, then installing node.js would be first step, then yarn (`npm install -g yarn`). Once both are available, clone this repo, edit [this file](https://github.com/keyboardio/chrysalis-bundle-keyboardio/blob/master/src/main/index.js#L35), and add the following line: `window.webContents.openDevTools();` (at around line 35, somewhere within the `createMainWindow` function)

Then `yarn install` followed by `yarn run start` will start Chrysalis in development mode, with devtools open. Switch to the console tab there, and see if there are any errors printed.

gavinhoward, 2018-12-31:
So, with devtools enabled, it works. I don't know if that helps. I can still give you the console output, if you want.

algernon, 2018-12-31:
It helps - it means there's something iffy going on with the AppImage. I have a few suspicions, will fire up an Ubuntu VM and see what I find. Thanks for the help os far!

gavinhoward, 2018-12-31:
You're welcome!

algernon, 2019-01-04:
HAH!

`export ELECTRON_ENABLE_LOGGING=1` is the trick, I learned, which revealed this:

```
[4279:0104/143728.375016:INFO:CONSOLE(88)] "Uncaught Error: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /tmp/.org.chromium.Chromium.7eKJEH)", source: /home/algernon/Downloads/C/squashfs-root/resources/app.asar/node_modules/bindings/bindings.js (88)
```

So I need to compile on a distribution with an older libc. And the reason the 0.1.0 release worked more widely is because I compiled it on an ancient Debian testing. Lovely. But at least I know how to work it around.

algernon, 2019-01-04:
I'm closing this issue, because the fix is to compile on an older system, which I will do for 0.3.0. Said version will also include a small change that will print a useful error to the terminal in cases like this, instead of failing silently with a white window.

closed 2019-01-04 by algernon

gavinhoward, 2019-01-04:
Sounds good!

PR #136 (2019-01-04): main: Catch the gpu-process-crashed event, and print a message
If, for some reason, the renderer process crashes, we can't display a proper error message in the browser window, so log one to the terminal with a few instructions about how to file a useful bug report about it.

This does not fix #112, but in the event of a similar bug occurring, we'll have more clue and more information to start with.
commit 234b7517 (2019-01-04): main: Catch the gpu-process-crashed event, and print a message

If, for some reason, the renderer process crashes, we can't display a proper
error message in the browser window, so log one to the terminal with a few
instructions about how to file a useful bug report about it.

This does not fix #112, but in the event of a similar bug occurring, we'll have
more clue and more information to start with.