keyboardio/Chrysalis #119: Selecting a key to edit, and pressing a key should select the pressed key

algernon, 2019-01-02:
To make the most common operations easier, instead of having to select a key, navigate over  to the key selector, select the desired key and click it, we could allow the user to click on a key, press an alpha-numeric on their keyboard, and have it selected automatically.

algernon, 2019-01-14:
With #201, this is now open again.

pm0u, 2020-05-16:
What exactly is required to make this a possiblity? for instance, grabbing a lowercase `a` yields `keyCode: 97, charCode: 97, code: "KeyA"`. it looks like the entry in the db is:

```
    {
      code: 4,
      labels: {
        primary: "A"
      }
    },
```

Should 97 be mapped to this? or should "KeyA"? will this vary by keyboard language?

Is there any context the the previous attempt/ revert that needs to be looked out for?

obra, 2020-05-18:
Doing this "really right" would, I think, want firmware interaction over
serial, so the keyboard can report which physical key got pressed, so that
if you, say, had two keys mapped to "a", it would light up the correct one.
That'd also let us switch the layer being displayed based on the hardware
state of the keyboard.

@algernon?

commit 727fdbf2 (2019-01-13): KeySelector: Select keys based on keys pressed

When the key selector is shown, listen for key-down events, and try to look up
the pressed key from the big table of codes. The lookup is done on the entire
table, not just the parts shown, and supports modifiers too.

Fixes #119.