keyboardio/Chrysalis #143: Colormap & Layout editors need a "Cancel pending changes" button

algernon, 2019-01-05:
When we made changes, but decide we don't want them, we need a cancel button for that. Right now, the workaround is to switch to another screen (not tab, screen) and back, but that's not intuitive. Cancel should re-init the screen, basically, and pull down the keyboard state again.

algernon, 2019-01-06:
We could put an `X` in front of the layer tabs if there are pending unsaved changes. I wouldn't put it on the right side, because there will be other stuff there (#149, or something similar).

obra, 2019-01-07:
Often I've seen that done with a pencil to the right of the text
(material's "create" icon). Though I was thinking we'd use the "block" icon
overlaid over the "create"

algernon, 2019-01-07:
I'll see what I can do about overlaying icons.

obra, 2019-01-07:
Overlaying icons is not a requirement!

Simply showing the icon at 50% opacity might be the right thing to do?
Actually, does material have a standard for "disabled icon". It does!
https://material-ui.com/style/icons/#svg-icons

algernon, 2019-01-07:
Wondering where to put the trash/cancel/whatever icon: I think we should flip the appbar into a contextual alternative when there are pending changes. This would prevent accidentally losing pending changes, since the page switcher menu would be inaccessible until either explicitly cancelling the changes, or saving them.

This is mostly in-line with (my interpretation of) the material guidelines, I think.

commit 98f7dc11 (2019-01-11): Implement a way to cancel pending changes

When there are pending changes in the Colormap or Layout editors, switch the
appbar to a context bar, with the main menu replaced with a cancel button. When
that icon is clicked, reload the data from the keyboard, effectively canceling
any pending changes.

Fixes #143.