keyboardio/Chrysalis #155: Allow colors on the keymap too

algernon, 2019-01-07:
We're at the point where we're encoding too much information into labels (DualUse, OneShot, etc). It would be helpful if we could use colors and/or textures to signal that a given key is special. We'd have a legend on the lower left side, or as tooltip.

See the [ErgoDox EZ configurator](https://configure.ergodox-ez.com) for inspiration.

TreTuna, 2019-01-22:
Want to look into adding this capability. First thing I can see is making a good list of all the different special keys we want to highlight in this fashion. 

Ones I can think of atm:
- Modifier combos? 
  - Hyper
  - Meh ( I have 4 variations of "meh")
  - other combos? (I personally have modifier keys for (mac) shift+Option, shift + control, control + command) 
- DualUse
- OneShot
- TapDance
- LayerLock
- LayerShift
- Macros
- Leader

Am I missing any?

algernon, 2019-01-22:
FWIW, some of the keymaps (the Model01's) already support background colors & key labels at the same time, although in a bit twisted way: if you give the component a `palette` and a `colormap`, it will use that, **and** the labels.

So in the case of the Model01, it's a case of passing a custom palette and colormap to the `<Layer>` component. We can hardcode the palette for now, and for the colormap, one idea I had in mind was to iterate over the keymap, find the group names for each key, and see if we have a color for that group.

Having something like this would do the trick:

```javascript
const groupToColorMap = {
 "Modifiers": 1,
 "OneShot modifiers": 2
};
```

Where the value is the index into the palette. If no group is in the map, use 0xf as the color (which would be white, I guess).

Does this make sense?

algernon, 2019-01-22:
Oh, and the logic of what to highlight and how, should be in Chrysalis, not in `@chrysalis-api/keymap`.

TreTuna, 2019-01-22:
Definitely got ya on the logic being in Chrysalis and not in keymap, that's exactly where my head is at with this. I just got a chance to look through the code and I believe I got what you're going towards here with the palette and colormap as well. That definitely seems like a good way to go, I'll dig in more when I get a chance.

algernon, 2019-02-20:
Technically, #322 allows colors on the keymap. But we use it for the colormap, not to color modifiers & the like. From the feedback I received, using it for that purpose would not be a useful change, and we'd have to place legends somewhere to further complicate things. So marking this as wontfix.

closed 2019-02-20 by algernon