keyboardio/Chrysalis #156: Notify the user when flashing fails due to a tool not available

algernon, 2019-01-07:
Some boards (ErgoDox & Atreus) are flashed with non-JS utilities, which at the moment need to be on PATH. We should tell the user this, and either point them to a place where they can download said utilities, or ship with apps for all the major platforms.

obra, 2019-01-07:
Bundling the binaries, if allowed, for OSX, Linux and Windows is my
preference.

algernon, 2019-01-07:
Should be, `teensy_loader_cli` is GPLv3, dfu-programmer is free software too, so we're good. Though, to be honest, seeing as they're all open source, I might just write a JavaScript library, and then we don't need to compile anything.

algernon, 2019-01-07:
Looked at the `teensy_loader_cli` sources - it's fairly straightforward. I can probably implement it in JS in a few hours.

algernon, 2019-01-07:
For 0.3.0, I'll go with having a note. We'll bundle/rewrite-in-JS for 0.4.0.

PR #172 (2019-01-11): FirmwareUpdate: If the hardware requires external tools, tell so
If the hardware we're about to flash requires an external tool, let the user know.

This fixes #156.

[image]
commit 3337b5de (2019-01-11): FirmwareUpdate: If the hardware requires external tools, tell so

If the hardware we're about to flash requires an external tool, let the user
know.

This fixes #156.
closed 2019-01-11 by obra