keyboardio/Chrysalis #210: Improve OneShot support

algernon, 2019-01-17:
DualUse is integrated into the key selector much better than OneShot is. To make a modifier or layer key oneshot, one has to select the type, then re-select the modifier/layer - not exactly friendly.

Instead, we should have another dropdown, or a switch below the key type selector, and move DualUse stuff there too.

algernon, 2019-02-04:
This will likely require another reorganization of the key selector component.

qayshp, 2019-02-05:
Would also be nice to be able to enable `ActiveModColorEffect` for `OSM`/`OSL` from here... but sorry, I'm not sure if that would require a firmware change too.

qayshp, 2019-02-05:
huh, just noticed that it is enabled... sorry! just not sure how to change the color :)

algernon, 2019-02-05:
> just not sure how to change the color :)

That's only possible by changing the firmware at the moment. If you can file an issue on Kaleidoscope, requesting that `ActiveModColorEffect` should support changing the color via Chrysalis, that'd be great, and I wouldn't forget to implement it (it's pretty easy to do so).

algernon, 2019-02-22:
Pushing this to 0.6.0 for now.

algernon, 2021-01-25:
With the Editor redesign in 0.8.0, this has been addressed: OneShot layers appear together with the other layer keys on the sidebar:

[image]
[image]

closed 2021-01-25 by algernon