keyboardio/Chrysalis #214: Device chooser dropdown should be styled as a dropdown

obra, 2019-01-17:
[image]

obra, 2019-01-17:
Also grey, I think? https://material.io/design/components/menus.html#exposed-dropdown-menu

algernon, 2019-01-17:
But it's not "exposed", not in its default state. The widget was modeled after [Selected Menus](https://material-ui.com/demos/menus/#selected-menus) in the library we use, but I can't find this in the Material Guidelines at the moment.

I'm not really fond of the grey background in this case... can't really explain why at the moment... I find it ok for things on the Settings page for example, but this feels different. The items are usually multi-line for one (second line being a port or "Unknown" in dimmer, smaller text).

obra, 2019-01-18:
Let's see how we do with just the triangle, then.

commit d1275b7b (2019-01-18): KeyboardSelect: Turn the select list into a real dropdown

Fixes #214.
closed 2019-01-18 by obra