keyboardio/Chrysalis #215: Move "Connect" next to the thing we're connecting to.

obra, 2019-01-17:
[image]

obra, 2019-01-17:
That's fine with me.

algernon, 2019-01-17:
Hmm... rearranging things like that makes the keyboard icon lonely up top in the center. It looked fine with the dropdown being centered too. With the dropdown on the left, connect on the right... not so much. I can either remove it, integrate it into the dropdown (or move it onto the same line, left of the dropdown), or leave it in place in which case we'll figure out what to do with it later.

obra, 2019-01-17:
Maybe put it to the left of the dropdown?

algernon, 2019-01-18:
This feels much better to me:

[image]
[image]

obra, 2019-01-18:
That's acceptable to me :)

closed 2019-01-18 by obra