keyboardio/Chrysalis #223: If I don't hold in prog when trying to update the firmware on the Model 01, the results are unreliable

obra, 2019-01-17:
Sometimes it gets stuck. Sometimes it goes back to the home screen with a little error in the corner that seems to vanish.

I think that if there's a failure we should -stay- on the flashing page? Maybe?

But the "Error flashing the firmware" message definitely shouldn't vanish. 

The Model 01 has two device IDs specifically so we can detect when we're in the bootloader vs in the regular firmware. It should be possible to detect a reboot that doesn't go into the bootloader and to not abort but instead try again?

algernon, 2019-01-18:
We can't reliably try again sadly. I've tried that route, and it was incredibly unreliable, there was too much state to juggle. I think the best we can do is display a different error message depending on whether we detected a bootloader or a normal keyboard connecting back after the reset.

And yes, the error should be sticky, and need human interaction to close.

algernon, 2019-01-19:
If this is reproducible (I always get taken back to the keyboard selection screen), can you grab a console log?

algernon, 2019-02-22:
Ping @obra, can you reproduce this?

algernon, 2019-03-08:
Can't repro, so moving it off of the 0.5.0 milestone.

algernon, 2020-04-29:
I've been pondering about this again, lately. I can't reproduce the getting stuck part, I always get taken back to the keyboard select screen. As for staying on the flash screen after an unsuccessful flash... that _might_ be possible now, after #508, because we can explicitly wait for a bootloader now, thus, can detect if it came up, and we might be able to reconnect if it did not.

As an extension, we can also likely detect the keyboard coming back up after a successful flash, and can stay on the same screen then, too.

Mind you, there's a caveat: if one plugs in another keyboard of the same type during the flashing process, we won't be able to tell which one of the two we should be reconnecting to. But that's such an edge case that we should be able to live with it, and document it if it ever comes up. Of course, we can work that around too, as part of #509: prior to flashing, we can save a bit or byte to EEPROM to  tell the keyboard we're about to flash it. When it comes back up, during probing, we can ask it to tell us if it's been recently flashed, and connect only if it were. (The bit is cleared when we ask for it, and after a minute if unasked)

@obra what do you think?

obra, 2020-04-29:
I do not think we need to worry about that edge case. Really. I'd run with the simpler implementation.

algernon, 2022-05-17:
Closing this, because the original issue is long fixed. Not having to reconnect after a successful flash is a different scenario, and #224 covers that.

closed 2022-05-17 by algernon