keyboardio/Chrysalis #224: If firmware flashing is successful, we currently end up on the "device chooser" page and need to disconnect/reconnect.

obra, 2019-01-18:
If the keyboard flashes successfully, we should do an automatic disconnect/reconnect and go to the layout editor UI

algernon, 2019-01-18:
Needing to disconnect is definitely a bug. Reconnect is a bit tricky, because we need to wait for the keyboard to boot back up.... I suppose I can make the update process wait for reboot too after flashing, in which case a reconnect should be much easier than if we try ASAP after flash being done.

For the record, I played with this before, and the results were disappointing - but I didn't wait for the reboot after flashing, and that's likely the key for reliability.

algernon, 2019-02-01:
We shouldn't need to disconnect now, but we do end up on the device chooser screen. For 0.4.0, I'm tempted to leave it that way, because reconnecting is a complicated matter.

algernon, 2019-02-22:
I'm not sure we can reliably do a reconnect... if multiple boards of the same device (eg, two Model01s) appear at roughly the same time, we won't know which one we need to reconnect to. An edge case, but still. I'd rather delegate the task to a human, even if that's a little bit more inconvenient.

*If* we had a unique keyboard ID we could reliably reconnect though. Something to consider in parallel to #323.

algernon, 2020-04-29:
[Here](https://github.com/keyboardio/Chrysalis/issues/223#issuecomment-621345820) is an idea about how we could semi-reliably do automatic reconnect.

algernon, 2022-05-19:
If #595 gets fixed, and we auto-connect if there's a single supported keyboard connected, that would cover a large part of this. Not completely, but that'd get us closer.

algernon, 2022-05-27:
With #797 merged, we now automatically reconnect when there's a single keyboard present. That's as close as we can reliably get, so I'm closing this issue.

closed 2022-05-27 by algernon