keyboardio/Chrysalis #225: Transparent keys shouldn't say "Trns", they should just be empty

obra, 2019-01-18:
[image]

"Trns" doesn't communicate a whole lot, and is also the default state of keys. Let's try just making transparent keys not say anything.

algernon, 2019-01-18:
What should blank (`Key_NoKey`) keys display as then?

One idea I had was to make it possible to have colors or textures on the keycaps, even in the layout editor (#155). That would allow us to have transparent keys blank and white, `NoKey` as blank with a grey-ish background.

obra, 2019-01-18:
"No Key" as XXX (at least for now) makes sense to me. Once we're at the
point of colors and things, we could grey it out. but not important for now.

algernon, 2019-01-18:
It's technically possible, all of the SVGs are set up to support both at the same time. It's just a matter of figuring out the appropriate colors, and adding a few lines to put the appropriate colors under keys that need it.

obra, 2019-01-18:
*nod*  No-key as greyed out would also be acceptable

TreTuna, 2019-01-22:
So, it sounds like this should be a change in https://github.com/keyboardio/chrysalis-api/tree/master/packages/chrysalis-keymap to mark NoKey as "XXX" and Transparent as "" (empty string). I can make this change, then I will look into working on https://github.com/keyboardio/Chrysalis/issues/155 to add some colors to the keys to distinguish different states.

obra, 2019-01-22:
That makes sense to me

algernon, 2019-01-23:
I have another bug to fix in `@chrysalis-api/keymap`, and will upload a release after that. Once Chrysalis updates its dependency, this issue should be fixed.

Thanks @TreTuna!

algernon, 2019-02-04:
Indeed, thanks for the reminder!

closed 2019-02-04 by algernon