keyboardio/Chrysalis #226: Copy should be "copy from" rather than "Copy to"

obra, 2019-01-18:
[image]

"Copy from" is closer to what the user is likely to want to do. "I have this layer. I want to make it like this other one"

Also, it makes you be on the screen with the layer being destroyed, which is much more likely to prevent the user from shooting themselves in the face.

algernon, 2019-01-18:
Same applies to the colormap editor then. It's been tough to choose which behaviour to implement, and my first instinct was "Copy to...". Having used it, I agree, it should be "Copy from..." instead.

obra, 2019-01-18:
*nod* It's never wrong to ask me for opinions in advance, but also, not all
that wrong to just try it one way ;)

PR #234 (2019-01-18): Editors: Copy layers from, not to
Copying layers from another one is a much more straightforward operation than copying to, primarily because we're on the layer to be destroyed when we begin. As such, there's less chance of shooting ourselves in the face.

Fixes #226.
commit 5c70ece9 (2019-01-18): Editors: Copy layers from, not to

Copying layers from another one is a much more straightforward operation than
copying to, primarily because we're on the layer to be destroyed when we begin.
As such, there's less chance of shooting ourselves in the face.

Fixes #226.
closed 2019-01-18 by obra