keyboardio/Chrysalis #228: When there are -no- custom eeprom layers defined, Chrysalis should default to copying in the hardcoded layouts

obra, 2019-01-18:
It should also switch the "default" layer to the copy of the old default layer and change the "switch to layer X" keys to the new copies.

If any of the eeprom layers are customized, it should simply not do any of this.

The default layer should be the one that gets focused.

The italics on the built-in layers aren't prominent enough for me. I think the text of those layers AND the whole keyboard should be #ccc or so (some slightly-disabled looking grey).

The intent of all of this is that a new user setting up Chrysalis should be able to:

* Download the app
* Open the app
* Be walked through a flashing flow 
* Be presented with their keyboard's default layout, with the ability to click on a single key,  change it, hit save, and have everything just work.

algernon, 2019-01-18:
Regarding "Switch to layer X" keys: I've been toying with another idea, layer switch keys that are relative to the default (keyboardio/Kaleidoscope#524). There's enough room in there to put these in, both for shifted and lock layer keys. The advantage of these would be that Chrysalis can simply copy them, and wouldn't need to understand the keymap while copying.

algernon, 2019-01-18:
Ow. Yes. Lets not do that then.

algernon, 2019-01-25:
Small update: what should we do with `keymap.map` and `keymap.roLayers`? If we move to the commands proposed above, these two old ones will become obsolete. Should we remove them? In that case, older versions of Chrysalis will not work with new firmware.

algernon, 2019-01-26:
I think I'll skip `keymap.useCustom`: it would need an extra byte of eeprom, unless we encode it into the default layer (but that's a gross layering violation). Instead, if default is higher than the last PROGMEM layer, we'll default to using EEPROM instead of PROGMEM.

obra, 2019-01-26:
that would break the use case for advanced users who want to have a mix of both and start on an eeprom layer :/

we could steal the top bit from the default layer byte...

algernon, 2019-01-26:
If they want to have both, they should use the `Mode::EXTEND` mode of EEPROMKeymap. This override thing would only be the case for `Mode::REPLACE`. Of course, this presents the problem that if someone would want to use both, they'd have to compile their own firmware, which is a bit meh.

We *could* steal the top bit, but that's controlled by EEPROMSettings. We can steal it, but it's a big layering violation. There's also a compatibility issue there: EEPROM starts off with 255, so the top bit would be set. However, if they set a default layer before, the byte will have its top bit unset.... on second thought, we already handle 255 as a special case, so could do here too.

In summary:

- first byte of EEPROM == 255: no default layer, use both progmem & eeprom.
- first byte of eeprom < 255: lower 7 bits are the layer; if top bit is unset, use both progmem & eeprom, if set, use eeprom only

We could call this bit `useEEPROMLayersOnly`.

We'll need a short name for the Focus command then (need to fit in 32 chars). While I have an uneasy feeling about it, `keymap.useCustom` is the closest. It'd just call `EEPROMSettings.useEEPROMLayersOnly(true)`.

algernon, 2019-01-27:
Stuff still looks the same, but I cleaned up a few things on the affected branches. Need to write docs, commit messages, and then I can submit PRs.

commit 9acca0b3 (2019-02-01): LayoutEditor: Update for the default/custom split

This is a major update to the LayoutEditor, which updates it to support the
default/custom split on the protocol level. While there, added the option to
hide default layers, and to set the `keymap.onlyCustom` flag too.

We also copy the default layers into EEPROM and toggle onlyCustom on if we see
that the EEPROM layers are all transparent.

Fixes #228.
PR #269 (2019-02-02): LayoutEditor: Update for the default/custom split
This is a major update to the LayoutEditor, which updates it to support the default/custom split on the protocol level. While there, added the option to hide default layers, and to set the `keymap.onlyCustom` flag too.

We also copy the default layers into EEPROM and toggle onlyCustom on if we see that the EEPROM layers are all transparent.

Fixes #228.

See screenshots [here](https://github.com/keyboardio/Chrysalis/issues/228#issuecomment-457874928)
closed 2019-02-02 by obra