keyboardio/Chrysalis #240: At some point, Chrysalis should be able to flash a Model 01 from the bootloader.

obra, 2019-01-18:
This is useful if the user builds bad firmware and we need to flash from recovery

algernon, 2019-01-19:
This is totally doable. Easiest way: have a Model01Bootloader device, with the bootloader PID, and flash it without reset.

This would, in practice, mean that Chrysalis would detect the keyboard as something that lacks features, go to the welcome screen, then update firmware, flash without reset, and boom, we can now select the Real Device.

A bit of a hack, but this is doable right now, without bigger changes to Chrysalis. To do it properly and support bootloader PID/VID pairs, we'd need to think a bit more.

obra, 2019-01-19:
Having the hack now seems like a win.

algernon, 2019-01-19:
I'll make it happen then.

PR #241 (2019-01-19): Add support for the Keyboardio Model01 in boot mode
By upgrading the hardware libraries, we have access to `Model01Bootloader`, a variant of the `Model01` descriptor. This allows Chrysalis to detect such a Model01, and allows us to flash it, without needing to boot a user firmware for Chrysalis to recognise it.

Fixes #240, and since the hardware library also bumped the time it waits for the bootloader to come up, this also fixes #238.
commit 36b74c86 (2019-01-19): Add support for the Keyboardio Model01 in boot mode

By upgrading the hardware libraries, we have access to `Model01Bootloader`, a
variant of the `Model01` descriptor. This allows Chrysalis to detect such a
Model01, and allows us to flash it, without needing to boot a user firmware for
Chrysalis to recognise it.

Fixes #240, and since the hardware library also bumped the time it waits for the
bootloader to come up, this also fixes #238.
closed 2019-01-19 by obra