keyboardio/Chrysalis #247: Only a blank screen if no internet

probonopd, 2019-01-19:
Only a blank screen if no internet:

[image]

Reference:
https://github.com/AppImage/appimage.github.io/pull/1196

algernon, 2019-01-19:
That's... interesting, because it shouldn't be pulling anything from the 'net. Can you try with a more recent version? From the linked issue it seems you're working with `0.0.5`, while the latest release is `0.3.1`.

algernon, 2019-01-21:
Just tried to reproduce with current master (also the 0.3.2 release), and no luck, started up fine. Mind you, 0.0.5 had.... issues. Among them, being compiled for a glibc that was too new, so it couldn't load even its starting screen. That resulted in a blank page, even with internet available.

I suppose that's what happened in this case too.

TreTuna, 2019-01-22:
I just wanted to add that I cannot replicate this blank screen either. Chrysalis works fine for me without an internet connection.

probonopd, 2019-01-22:
OK, thank you. Possibly it is an issue with our Firejail sandbox setup. Does the application need any special permissions?