keyboardio/Chrysalis #251: Window to keymap resizing needs better consistency

TreTuna, 2019-01-22:
When resizing the main window, the UI of the keymap does not resize consistently. It starts off filling about 90% of the window, then shrinks drastically, then goes back to the previous size, then finally gets slightly larger. 

We should have this resize consistently to a maximum size and then stop. Perhaps working with the percentage of the screen vs media queries in order to avoid jumpy resizing.

Link to a video of the offending behavior: https://www.dropbox.com/s/jc9ee3vd6drro70/Screen%20Recording%202019-01-21%20at%205.07.36%20PM.mov?dl=0

algernon, 2019-01-22:
There are some media queries in place, but they're kinda... wonky, indeed.

algernon, 2019-02-06:
#267 is merged, closing this.

closed 2019-02-06 by algernon