keyboardio/Chrysalis #252: Dark theme

TreTuna, 2019-01-22:
Create at least one more theme that swaps over to using darker colors. For those of us that cannot stand the light!

We could add a switch in the settings to toggle between the two. There is also a setting you can use in Electron to detect of a Mac is set to "dark mode" and we could use that for Mac users to automatically set their default to match their computer.

TreTuna, 2019-01-23:
I'm just adding this link here for reference when I (or someone else) looks into this more. This is how you automatically update the theme in Electron based on if a user with MacOS Mojave is in Dark Mode or not. https://electronjs.org/docs/tutorial/mojave-dark-mode-guide

infopowerbroker, 2019-01-26:
in Windows 10, it looks like the dark mode can be found by reading the following registry key: HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Themes\Personalize\AppsUseLightTheme  (dark =0)

algernon, 2019-01-26:
[node-regedit](https://github.com/ironsource/node-regedit) or [windows-registry](https://www.npmjs.com/package/windows-registry) could be used to look at the registry value, I imagine.

TreTuna, 2019-03-17:
I have created support for manual switching to the Material UI dark theme. However, the styles themselves should be tweaked in the theme at some point soon as there are some things that are not ideal (the black text on a grey background as per your screenshot above). Perhaps a rework of both light and dark themes are probably in order soon anyway so it doesn't look so stock Material UI. 

I think that adding support for this to pull from the user system settings should probably be an entirely different issue. I only have a Mac to test that out on, so I would not be able to test the Windows or Linux builds and support for that.

TreTuna, 2019-03-17:
https://github.com/keyboardio/Chrysalis/pull/355

hnrklssn, 2020-08-16:
@algernon This has been implemented now right?

Rom1deTroyes, 2020-08-24:
it's working pretty well ATM : there is for sure some improvements to do, but nothing really urgent  (more cosmetic things).
You can try it choosing `Dark theme` in `Preferences` and send your feedback here ;-)

I start to play with the code, but my level is not already sufficient to work on this :-(

algernon, 2020-10-09:
We're using the theme palette for all custom colors, so we should be good in almost all cases, and Dark mode should be pretty usable at this point.

As such, I'm closing this issue. If there are any improvements to be made, lets open issues with smaller scopes for those.

closed 2020-10-09 by algernon