keyboardio/Chrysalis #253: Warn user their firmware doesn't have the colormap plugin instead of hiding the colormap editor

TreTuna, 2019-01-22:
Currently, if I do not have the colormap plugin installed on my firmware, the colormap editor is hidden. Instead, we should not hide the colormap editor, but simply disable it and show a message that tells a user that they need to install colormap to enable that feature of Chrysalis.

infopowerbroker, 2019-01-25:
Is it possible to have the "experimental" firmware include colormap? That would allow Chrysalis to shine with full features without some noob like me needing to compile a firmware from scratch. ( I didn't update my firmware until today because doing it from scratch was just a bit too daunting for me, but it was super simple in the GUI)

TreTuna, 2019-01-25:
Seems reasonable, could make a PR to that sometime.

P.S. love that avatar, I have a canvas of that hanging in my studio. 😁

algernon, 2019-01-25:
The [experimental firmware](https://github.com/keyboardio/Chrysalis-Firmware-Bundle/blob/master/Keyboardio/Model01/experimental/Model01.ino) should include colormap. If it doesn't, then I screwed something up.

TreTuna, 2019-01-25:
I just flashed over to it and am not able to see the colormap... So, yeah..... Something's going on there.

infopowerbroker, 2019-01-25:
I just flashed mine to experimental 0.3.3, and now colormap is present. Thanks!

algernon, 2019-02-22:
How should we handle this in light of the merged editor? I do not think a warning is a good choice, because it is entirely possible that one does not wish to have the colormap plugin enabled at all, and we should respect that.

I think the best course would be better documentation, a Wiki page with screenshots and a tour of the UI - once we stop redoing major parts every release.

algernon, 2019-02-22:
@TreTuna, @obra any opinions, ideas with regards to the above comment?

obra, 2019-02-27:
If their firmware doesn't have colormap support, we shouldn't be showing
the color support.

Longer-term, perhaps somewhere in the keyboard configuration page, show a
list of all detected plugins, as well as features in Chrysalis that we
haven't detected support for?
ᐧ

TreTuna, 2019-02-27:
I don't like just hiding it because then people might wonder where it is or why it seems they can't adjust colors with it since we advertise that as a feature. I would disable the button and have a tooltip that says something like "The current firmware does not support this feature. Please add the Colormap plug-in to enable."

TreTuna, 2019-02-27:
I really like the idea of a list of currently detected plugins somewhere.

algernon, 2019-02-27:
> I don't like just hiding it because then people might wonder where it is or why it seems they can't adjust colors with it since we advertise that as a feature. I would disable the button and have a [...]

From another point of view: there are many boards Chrysalis supports that do not have colormap support at all. Showing the disabled button from them would be wrong. Showing the button based on what device we're connected to feels wrong too.

A list of detected plugins would  address the issue better, I believe. (And it will be possible as a side-effect of the EEPROM rework I have in progress)

obra, 2019-02-27:
As an aside, will adding in the Colormap plugin to the Model 01's firmware
now scramble the EEPROM data if it wasn't there before?

If not, we ought to just roll it into the non-experimental Model 01
firmware, making this a non-issue for most folks.
ᐧ

algernon, 2019-02-27:
If added carefully, no, it won't. I'll submit a PR later today, if all goes well. (The factory firmware is a bit short on PROGMEM space, but Colormap should still fit I think)

TreTuna, 2019-04-06:
I'm closing this issue as it isn't the route we want to take. With the thoughts here it seems better to open one at some point for a feature to list detected plugins.

closed 2019-04-06 by TreTuna