keyboardio/Chrysalis #257: Allow unknown key codes to be augmented with modifiers

hnrklssn, 2019-01-22:
After a bit of digging I've found that the <>| key in the standard Swedish keyboard layout doesn't exist in the US layout, but is still accessible through keycode 100 in Chrysalis. Currently, however, it is not possible to map 100, 100+shift, and 100+altgr to different keys, as the modifier checkboxes are not available for the custom keycode selector.

algernon, 2019-01-23:
We can't allow modifiers for all unknown keycodes, because modifier augment only works for keycodes smaller than 256. What we can do, is add the missing keycodes to the keymap database, so they won't be unknown anymore. 100, for example, is "Non-US backslash and pipe".

obra, 2019-01-23:
Am I remembering wrong or aren't all "Keyboard" HID codes under 255 anyay?
ᐧ

algernon, 2019-01-23:
They are, but not all of them are in our database yet, so some of them appear as "unknown keycode".

obra, 2019-01-23:
Right. But what I mean is that anything that's a "keyboard key" keycode
should allow modifier augment, even if it's unknown.

ᐧ

algernon, 2019-02-22:
In principle, I agree that anything below 255 should allow augmentation. However, doing so would mean I have to reorganize the code considerably. It's easier to add all valid keycodes to the database, so I'll be doing that instead.

algernon, 2019-03-08:
Postponing this, as there are close to 150 keycodes currently missing from our database, many of them almost completely useless.

Missing codes are:

1, 2, 3, 50, 83, 88, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155, 156, 157, 158, 159, 160, 161, 162, 163, 164, 165, 166, 167, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, 200, 201, 202, 203, 204, 205, 206, 207, 208, 209, 210, 211, 212, 213, 214, 215, 216, 217, 218, 219, 220, 221, 222, 223, 232, 233, 234, 235, 236, 237, 238, 239, 240, 241, 242, 243, 244, 245, 246, 247, 248, 249, 250, 251, 252, 253, 254, 255

Some of these (1, 2, 3 for example) shouldn't be in the DB, or be hidden or something. Possibly others too, I don't have the resources to go over all of them at the moment. Keycode 100 has been added already, and we'll gradually add the rest. Just not right now.

algernon, 2021-01-25:
Closing this - most (if not all) - useful keycodes are in the db now. If any is still missing, we'll open a separate issue about those.

closed 2021-01-25 by algernon