keyboardio/Chrysalis #259: Unable to edit layers with  EEPROM laready installedc

compostbrain, 2019-01-24:
[image]

TreTuna, 2019-01-25:
If you look at the layer tabs at the top you will see that the first three are marked with (RO), which means read-only. Your EEPROM layers start at layer 3. We're working on making this beginning workflow better, but for the moment I've found the best way to go about the initial setup is to do the following steps:

Copy layer 0 to layer 3 and mark this as the default layer
Copy layer 1 to layer 4
Copy layer 2 to layer 5
Change any "ShiftTo" and "LockTo" keys to mark new layers (1 becomes 4 and 2 becomes 5)
Sync this to your keyboard
Unplug and replug it in for the default layer change to take hold

RickCogley, 2019-01-25:
I'm not one of the devs, but isn't the RO after the layer you have selected indicating it is Read Only?

algernon, 2019-01-25:
> Unplug and replug it in for the default layer change to take hold

With the new default & experimental firmware this step is no longer needed, because Kaleidoscope switches to the default layer when it is set, without replugging.

compostbrain, 2019-01-25:
Thanks for the explanation. It all makes sense now that I know what the (RO) means.

algernon, 2019-02-22:
I'm closing this in favour of #297, which better describes what we need to do.

closed 2019-02-22 by algernon