keyboardio/Chrysalis #260: Which version, color map editor

RickCogley, 2019-01-25:
Hi - thank you for your work. Just wondering a couple things:

* how can I tell what version of Chrysalis I have? Is it safe to assume that the Chrysalis listed in the firmware update section has that listed?
* how can I access the color map editor? It doesn't appear in the menu whether I load the default or the experimental firmware. 

Thanks!

macOS Mojave 10.14.2 (18C54)

TreTuna, 2019-01-25:
Looking into this right now. The experimental firmware _should_ have colormap installed on it already. At least it appears to in the [repo](https://github.com/keyboardio/Chrysalis-Firmware-Bundle/blob/master/Keyboardio/Model01/experimental/Model01.ino). However, I just flashed to it from Chrysalis and am having the same issue.

algernon, 2019-01-25:
Ow. I think I know why. There appears to be a bug in Chrysalis, wherein it will flash the *default* firmware if there is no custom firmware selected. So flashing the experimental one does not currently work.

Which explains why you don't see the colormap - it's not in the *default* firmware.

algernon, 2019-01-25:
> how can I tell what version of Chrysalis I have? Is it safe to assume that the Chrysalis listed in the firmware update section has that listed?

Yes, the firmware update section lists the version of the running Chrysalis. We should add an About or Help screen nevertheless, where this information is more readily available.

RickCogley, 2019-01-25:
thanks, noted on how to tell what version, and I'll test as soon as there is something to test. Looking forward to trying it, and I'm just glad it isn't me.

PR #262 (2019-01-25): This adds the Chrysalis version to the sidebar
Fixes #260.

[image]
commit cfb028fc (2019-01-25): This adds the Chrysalis version to the sidebar

Fixes #260.
closed 2019-01-25 by obra