keyboardio/Chrysalis #263: Trouble after upgrade to 0.3.3

RickCogley, 2019-01-26:
Hi, thanks for releasing [0.3.3](https://github.com/keyboardio/Chrysalis/releases/tag/chrysalis-0.3.3). I am able now to flash with "experimental" but, there are a few other problems:

* when I upgraded and flashed with experimental, when the model 01 came back online, the layers 3, 4 and 5 that I had set up were "garbled" for lack of a better word. I'm not sure if there was any pattern but, it appeared that the keys were all kind of mixed up. I re-copied the base layers to three layers and re did my settings. 
* now I can select the color map editor, but, it does not appear to actually do anything. I figured out how to assign a color to a key, and tried it in a couple of my layers but, nothing ever seems to happen. 
* now I get a lot of notification errors on iOS, saying something like "unplug the usb device that is using too much power" or some such, and generally the keyboard disconnects a lot now when I am in Chrysalis. 
* when I hit the led button to cycle the led patterns, it's no longer a smooth transition. There is a delay loading each and sometimes, it fails altogether or there is a single flash then nothing. 

Not sure what all this means.

algernon, 2019-01-26:
> it appeared that the keys were all kind of mixed up.

Is it possible that the keys were shifted a few positions? One cause of this might be if I screwed up the plugin order in the experimental firmware... if Colormap inits before EEPROMKeymap, then it will use parts of the EEPROM that were assigned to the keymap with the default firmware. I'll check this myself.

> now I can select the color map editor, but, it does not appear to actually do anything.

You need to switch to the Colormap LED mode. I guess for less surprises, the experimental firmware should only contain LEDOff and Colormap, so pressing the LED key would toggle between this two.

> now I get a lot of notification errors on iOS, saying something like "unplug the usb device that is using too much power" 

Does this happen when you have colors active on the keyboard? The LEDs tend to use quite a bit of power, especially when set to a brighter color.

> when I hit the led button to cycle the led patterns, it's no longer a smooth transition. There is a delay loading each and sometimes, it fails altogether or there is a single flash then nothing.

No clue about this yet.

Thanks for the feedback, I'll see what I can figure out.

RickCogley, 2019-01-26:
Hi - I reflashed back to standard to try to get rid of the usb memory errors that were not occurring yesterday. 

> Is it possible that the keys were shifted a few positions? 

Here is a screenshot:

[image]

> You need to switch to the Colormap LED mode. 

I could not find this in the layout editor at all...

> Does this happen when you have colors active on the keyboard? 

Hard to say. Yesterday when I was in normal firmware mode installed w/ the last version of Chrysalis, it was not happening at all, despite having the LEDs on all day. I don't think I selected anything particularly special...

algernon, 2019-01-26:
Hm, that doesn't appear to be shifted, but truly mixed up. I'll look into it, thanks!

algernon, 2019-02-22:
I've looked into it now, and I believe this was caused by the Colormap plugin squeezing its data in front of the keymap with the experimental firmware. This has been fixed since, but we do not (and can't reliably) fix old EEPROM layouts.

To fix things, the best solution right now is to wait for #326 to land, clear the EEPROM, and start afresh. (Sorry!)

Since there's nothing we can automatically do at this time, I'm inclined to close this issue.

TreTuna, 2019-04-06:
Closing this issue as we are far beyond v0.3.3 now, newest version gives options to work around this, and the OP hasn't responded to this in a long time. Happy to reopen if the OP does come around and still having these issues.

closed 2019-04-06 by TreTuna