keyboardio/Chrysalis #270: The experimental firmware should have compatible EEPROM layout

algernon, 2019-02-01:
Right now, the Model01 experimental firmware has an EEPROM layout that is not compatible with the default one: there's a palette before the keymap. Anything extra in the experimental firmware should come after the keymaps.

obra, 2019-02-02:
I feel like this will be an ongoing problem. Should we have a way to force
ordering? I feel like our lives will be easier if EEPROM is easy to keep
stable between builds.
ᐧ

algernon, 2019-02-02:
It shouldn't be much of a problem, once I add comments. Just have to remember to add stuff at the end and not in between, and we're good.

keyboardio/Kaleidoscope#344 will be some help, because we can then save the eeprom layout in Chrysalis, and if we see the same board again, with a different layout, we can either automatically rearrange if possible, or offer the user the choice.

algernon, 2019-02-03:
Marking it pending, keyboardio/Kaleidoscope#557 has a fix, we just need to rebuild the firmware we ship with Chrysalis.

algernon, 2019-02-03:
This is now waiting for keyboardio/Model01-Firmware#77. Once merged, I'll build new firmware.

obra, 2019-02-04:
That's merged.
ᐧ

PR #279 (2019-02-04): Update all the shipped firmware
Built from Chrysalis-Firmware-Bundle-0.3.1 (5b4c274), using Kaleidoscope master (733707c8).

Fixes #270.
commit c968cf42 (2019-02-04): Update all the shipped firmware

Built from Chrysalis-Firmware-Bundle-0.3.1 (5b4c274), using Kaleidoscope
master (733707c8).

Fixes #270.
closed 2019-02-04 by obra