keyboardio/Chrysalis #276: Inform users when they don't have enough permissions to flash firmware

undu, 2019-02-03:
When using Linux, if the user does not have permissions to write the serial port Chrysalis shows a generic message about an error when trying to flash the Model01.

A message with detailed information would be appreciated, as well as instructions on how to fix it (or even automate it)

In my case that was fixed by adding the user to the group that has access to the tty device and restaring the system:

```bash
$ stat /dev/ttyACM0
  File: /dev/ttyACM0
  Size: 0               Blocks: 0          IO Block: 4096   character special file
Device: 6h/6d   Inode: 16075       Links: 1     Device type: a6,0
Access: (0660/crw-rw----)  Uid: (    0/    root)   Gid: (  987/    uucp)
Access: 2019-02-03 09:17:45.108451175 +0000
Modify: 2019-02-03 09:17:45.108451175 +0000
Change: 2019-02-03 09:17:45.108451175 +0000
 Birth: -

```
then 
```bash
usermod -a -G uucp $(whoami)
```

algernon, 2019-02-22:
This is something that will have to be in the flasher libraries. It's not terribly hard to implement, but needs a bit of care.

obra, 2019-07-08:
FWIW, this is biting live customers. We need to, at the very least, improve the docs around this for the next release.

algernon, 2019-07-15:
I _think_ I have a reasonable solution for this, at least for Linux, possibly OSX too. We can also adjust the error message to have a link to a wiki entry about permissions. I think that link would be the most straightforward way to add documentation.

obra, 2019-07-15:
That seems like a fine next step.
ᐧ

algernon, 2020-04-29:
As it turns out, #500 was a duplicate of this, and has been fixed by #502.

closed 2020-04-29 by algernon