keyboardio/Chrysalis #290: Device setting and app settings should be sepaprated

obra, 2019-02-05:
Device settings and app settings are conceptually different. The device settings travel with the keyboard, while the app settings stick with Chrysalis.

Device settings should be on their own screen, possibly called "Device Settings" if we do that, -maybe- Chrysalis settings get called "Preferences"

The Device settings screen -may- be a more natural place to have "update firmware", "backup firmware", etc, than the side menu. but maybe not.

TreTuna, 2019-02-05:
I could see a "Keyboard/Device Settings" option in the top section of the menu with the editors and firmware options. 

I don't know if I would remove Firmware from the menu. We are making adjustments to the plugins and firmware enough right now that I think it warrants its place. However, I can see at some point turning from "Firmware Update" to just "Firmware" and within there having sections to Update, Export, and perhaps at some point "Modify" for things like adding plugins (though I know that would be a long way off if it ever happens since it would involve compiling firmware with Chrysalis).

obra, 2019-02-05:
Sure.
ᐧ

algernon, 2019-02-05:
I'm all on board with @TreTuna's idea. We can have subtitles in the sidebar too, so we can have things like:

```text
Keyboard
  Layout editor
  Colormap editor
  Settings
-------
Chrysalis
  Select another keyboard
  Preferences
   -----
   Discord
   etc
```

commit 25d716ec (2019-02-06): Split out keyboard settings from Preferences

Introduce a new menu, "Keyboard settings", splitting it out of "Preferences".

Fixes #290.
closed 2019-02-06 by obra