keyboardio/Chrysalis #302: Blank/white window after switching between color and keyboard layout

fluffy-critter, 2019-02-07:
While playing with Chrysalis I switched from the colormap editor to the layout editor, and the screen went blank. Log attached.

[-1549503166875.log](https://github.com/keyboardio/Chrysalis/files/2838981/-1549503166875.log)

fluffy-critter, 2019-02-07:
Got what I think is a related crash from clicking outside of the hamburger menu to dismiss it. 
[-1549514529157.log](https://github.com/keyboardio/Chrysalis/files/2839336/-1549514529157.log)

algernon, 2019-02-07:
I thought both of these would be fixed on master (by f6517e9f0864c137411049166c59b6bbf9e16a39), but apparently, there's still an issue somewhere. Thanks for the report!

algernon, 2019-02-07:
This is hopefully addressed by #304.

commit dfc67df5 (2019-02-07): When navigating, always await for it to finish

This should fix the remaining issues where we navigate somewhere that depends on
focus being in one state, but isn't yet.

Fixes #302.
PR #304 (2019-02-07): When navigating, always await for it to finish
This should fix the remaining issues where we navigate somewhere that depends on focus being in one state, but isn't yet.

Fixes #302.
closed 2019-02-07 by obra