keyboardio/Chrysalis #303: Please show the key map on top of the colormap editor

fluffy-critter, 2019-02-07:
When I'm editing a colormap it's probably because I want to set the colors on a per-key-type basis. It'd be cool if the actual key labels were visible on the colormap editor (maybe greyed out to indicate they're not editable and outlined in white so they get forced contrast with the background color).

TreTuna, 2019-02-07:
Hmm... I like this. Makes a lot of sense as well.

obra, 2019-02-07:
Yes. This makes a lot of sense

fluffy-critter, 2019-02-07:
Oh, and I'd also love it if there could be an "unset"/"passthrough" color so that in the case of layer stacking it just uses whatever would correspond with the key under it.

algernon, 2019-02-07:
> Oh, and I'd also love it if there could be an "unset"/"passthrough" color so that in the case of layer stacking it just uses whatever would correspond with the key under it.

The Colormap plugin used to have a transparent color support, but it was very slow, and was [removed](https://github.com/keyboardio/Kaleidoscope-LED-Palette-Theme/commit/4a494c686bb26538140798178033b6b63ff0745f). It might be possible to do it better now, I'll file an issue on Kaleidoscope.

commit a031cb88 (2019-02-20): Merge and redesign the layout & colormap editors

Instead of having a separate layout- and colormap editor, provide both on the
same screen. To achieve this, we needed to compact the secondary toolbar, thus
the layer tabs were turned into a dropdown instead. Chosing between keymap and
colormap editing can be done with a new pair of toggle buttons. The colormap
stuff is not offered if the board does not support colormap.

While there, the "..." (more) menu on the right side of the secondary toolbar
has been unpacked, the copy & clear functions were lifted to the main toolbar,
as icons. Clearing [... cut]
PR #322 (2019-02-20): Merged layout & colormap editor
Instead of having a separate layout- and colormap editor, provide both on the same screen. To achieve this, we needed to compact the secondary toolbar, thus the layer tabs were turned into a dropdown instead. Chosing between keymap and colormap editing can be done with a new pair of toggle buttons. The colormap stuff is not offered if the board does not support colormap.

While there, the "..." (more) menu on the right side of the secondary toolbar has been unpacked, the copy & clear functions were lifted to the main toolbar, as icons. Clearing now prompts for a [... cut]
closed 2019-02-20 by obra