keyboardio/Chrysalis #313: Chrysalis weirdness

0spinboson, 2019-02-09:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash a kaleidoscope-created firmware with 5 (?) layers to the Model01.
2. Flash Chrysalis-experimental
3. look at existing layers: 0-2 are from chrysalis, 3 and 4 are, in order, layers 1 and 0 from the old fw.
4. copy layer 1 to layer 3, which makes Chrysalis enter a kind of editing mode (?) Notice that layer 4 has suddenly changed completely from what was shown before.

**Expected behavior**
Layer 4 to contain the same layout it does when you exit editing mode.

**Screenshots**
Will add these in a bit, after switching OS.

**Desktop (please complete the following information):**
Win10, next to last build, Chrysalis .4+52

algernon, 2019-03-08:
Is this still reproducible? I _think_ the main problem here was the state of EEPROM, which can now be cleared, to start afresh.

TreTuna, 2019-04-06:
@0spinboson is this still an issue on v0.5.0+? I was not able to reproduce.

algernon, 2021-01-25:
This appears to be an issue with having hardcoded & eeprom layers mixed together being terribly confusing. As of 0.8.0, we no longer support such a mix, so this should no longer be a cause of confusion. As such, I'm closing this.

closed 2021-01-25 by algernon