keyboardio/Chrysalis #315: Long list of keys hide the dualUse switches

pdcawley, 2019-02-09:
If the list of keys in a category is long enough to wrap onto two lines, then the dual use toggle overflows the display and can't be seen.

**To Reproduce**

1. Go to 'Layout Editor'
2. Click on a letter key
3. At the default size on OS X, the list of possible keys wraps onto two lines and the switch labelled `Modifier when held, normal key otherwise` is clipped

**Expected behavior**
I expected to be able to enable dual use on letter keys. I only realised this was possible when I maximised the window. If the window is too narrow for all the options, I'd expect to see a scroll bar in the key selection area (or whatever the heck it's called).

**Screenshots**

* Two lines of keys:

[image]

* Wide enough for one switch to appear

[image]

* Ooh! I didn't even know about this option!

[image]

**Desktop (please complete the following information):**
 - OS: MacOS 10.14
 - Chrysalis Version: 0.4.0 AND master

TreTuna, 2019-03-21:
Mentioned this in Discord, however, I played around with moving things around and resizing the KeySelector and am not getting something I'm really liking with it yet. We may have to rethink the way these are incorporated.

TreTuna, 2019-03-21:
Just commenting here that this is related to https://github.com/keyboardio/Chrysalis/issues/210

mattvenn, 2019-05-14:
Perhaps I'm overlooking something, but why not just make the window bigger and the keymap smaller?

algernon, 2021-01-25:
We have redesigned the Editor, so this is no longer applicable, the new editor does not have this problem.

closed 2021-01-25 by algernon