keyboardio/Chrysalis #319: numlock layer colormap not showing in editor

qayshp, 2019-02-12:
**Describe the bug**
When enabling numlock layer, my keyboard colors change accordingly, which is great! I don't, however, see the colormap for this in the colormap editor for either the Read Only or custom numlock layers.

**To Reproduce**
Steps to reproduce the behavior:
1. Colormap editor
2. Click on a numlock layer
3. notice that the colormap is still all the default

**Expected behavior**
See the colormap showing that the right half numpad pattern keys would be red.

**Screenshots**
[image]
[image]

Nothing special showing in the log, but let me know if there's anything you need from it.

**Desktop (please complete the following information):**
 - OS: macOS 10.14.3
 - Chrysalis Version: 0.4.0+53

**Additional context**
N/A

algernon, 2019-02-22:
This is kind of "expected" behaviour. The colormap editor works with the colors provided by the `Colormap` plugin. Any other plugin that changes colors one way or the other, are not reflected in the editor. We currently do not have any plans to change this, but this should be much better documented to avoid this kind of surprises and confusion.

mattmc3, 2019-08-25:
What if we want the Chrysalis color map to take precedence of other plugins? Any way to do that? I'm experiencing a similar issue where I am using the butterfly key to switch from QWERTY to lock to a Colemak layer and all my remaps light up red which is really not ideal. Any fn key layer colors also don't show.

nickhandel, 2020-03-05:
I've just discovered this as well - out-of-the-box, I can set the Layer 1 colours to anything I like and it doesn't take effect.  From a dumb end user point-of-view, is there no way of changing this without getting into customising the firmware?

qayshp, 2021-04-09:
I understand that showing what keys/colors a plugin changes would be overkill.

Having a way to toggle stock firmware plugins without building a sketch would be a nice-to-have.

alerque, 2022-11-10:
My Model 100 arrived today. I'm pretty stoked, they keyboard is pretty nice. The first issue I ran into was the reasonably well documented relating to the stock firmware not reporting a version and thus needing a modified procedure to update.

The second issue I ran into was this one. I hacked on my layouts for a long time trying to bring them up to speed with my previous Kinesis setup. Then I started color coding keys into groups. Everything went fine except the numlock layer. No matter what I tried the colors just would not apply. Sometimes 1 key would cooperate, usually the keys I defined had colors (as opposed to being undefined) but never would they obey my settings.

Finally reading through this issue I found a workaround: copy layer 1 to an unused layer, then change the numlock modified to move to that layer. Now the colormaps work just fine as well as the keys.

This is unexpected and still kind of inexplicable. Why is layer 1 magic in a way that allows you to program the layer as usual but not the colors?

obra, 2024-11-11:
I'm closing this out in favor of #430 as a primary tracking issue for the Layer 1 / numlock colors issue.

closed 2024-11-11 by obra