keyboardio/Chrysalis #323: Improved keyboard detection

algernon, 2019-02-20:
**Is your feature request related to a problem? Please describe.**

We currently detect keyboards by their VID+PID pair, which works fine for boards that have unique pairs, but a lot of the DIY boards do not, and they reuse existing pairs. At the moment, we offer all matching boards in the selector, and delegate the choice to the end-user. We can - and should - do better than that.

**Describe the solution you'd like**

We need a way to uniquely detect a keyboard. There's not much in the USB descriptors we can use, and tieing ourselves to USB even further is not forward looking, so a Focus command that'd tell us the keyboard ID or name would do wonders.

When detecting keyboards, if it has an USB port, and matches one of our VID+PID pairs (we'd still keep that, to limit which devices we probe), we'd probe the device by asking for its name. If a name is found, we offer that board in the selector. If not, we fall back to the current behaviour of delegating to the end-user. In that case, we should make it clearer that we do not know for certain which keyboard a device corresponds to. An icon and a tooltip might do, perhaps.

**Describe alternatives you've considered**

- Considered looking at the USB vendor & product *strings* along with the IDs, but that ties us to USB too much. It might be a shorter-term option, or as an additional source of info, though.
- Considered a Focus command that returns an ID instead of a name, but then we need a global registry, which is something I'd like to avoid.

algernon, 2022-05-19:
Another thing to consider, where improved detection could be useful: automatically reconnecting after a flash or reboot. Ideally, we'd need to know that the USB device that came back is the same device we want to connect to. A serial or something similar would do the trick: we can connect, check, disconnect if not matching. That way we could have any number of the same type of keyboard plugged in, and we'd still reconnect to the right one.

obra, 2022-05-19:
Unfortunately for avr based boards, we don't have a real serial number.

algernon, 2022-05-19:
We don't need to have one. We can fake one with a small plugin + Chrysalis on first connect.

Rom1deTroyes, 2022-05-19:
Store a random serial number in the eeprom before disconnect and check for it on re-connect ?

kaleidoscope version + chrysalis version + datetime is simple to compute and, unless you are flashing a batch of keyboards at the same time, seems reliable enough.
You can erase this to reclaim space after the re-connection.

obra, 2025-06-01:
No supported keyboard needs this.

closed 2025-06-01 by obra