keyboardio/Chrysalis #324: Configure timeout for "modifier when held, normal key otherwise"

szymonm, 2019-02-22:
**Is your feature request related to a problem? Please describe.**
Configure timeout for "modifier when held, normal key otherwise". 

**Describe the solution you'd like**
The default latency is much too long for me. I would like to have it in "keyboard settings" tab. Can be global for all keys.

**Describe alternatives you've considered**
One can use karabiner for that feature on Mac.

**Additional context**
In Karabiner I used to have 500ms.

algernon, 2019-02-22:
This will require a bit of firmware support. The plugin in question (`Qukeys`) does not support setting a timeout via EEPROM yet. Once we have that, we can add the setting to the "Keyboard settings" screen.

qayshp, 2019-03-13:
Great suggestion!
It would also be helpful to have OneShot timeouts (both of them possibly?) settable.
Maybe worth creating another issue or adding to https://github.com/keyboardio/Chrysalis/issues/210

gedankenexperimenter, 2021-06-02:
> **Describe the solution you'd like**
> The default latency is much too long for me. I would like to have it in "keyboard settings" tab. Can be global for all keys.
> 
> **Additional context**
> In Karabiner I used to have 500ms.

Apologies for only seeing this now (I haven't kept up with Chrysalis at all).  These two statements are perplexing to me, and if you're still interested and able to describe the problem, @szymonm, I'd like to know more.  The default hold timeout for Qukeys is 250ms—half as long as the timeout you say you used in Karabiner.  Also, that default only applies for qukeys held alone, without pressing and releasing another key.  It's not necessary to hold the key down and wait for the modifier before pressing the key that it will apply to, because Qukeys delays all events until it knows which key was released first, then immediately sends all the events to the host. You can chord it very quickly with the modified key, and you might notice a slight delay in the characters appearing on the screen, because some of them will appear when a key is released, rather than pressed, but that doesn't mean you need to slow down your typing to use it.

mamapitufo, 2022-03-22:
I'm still interested in this feature.

I currently have the `d` and `f` keys (regular QWERTY layout) set to switch to different layers when I hold them. The problem I have is that if I type "diff" or "add" at my regular typing speed the second "f" or "d" won't be registered. It would be great to have the ability to set the timeout.

gedankenexperimenter, 2022-03-22:
> I currently have the d and f keys (regular QWERTY layout) set to switch
to different layers when I hold them. The problem I have is that if I type
"diff" or "add" at my regular typing speed the second "f" or "d" won't be
registered. It would be great to have the ability to set the timeout.

This sounds like an issue with the timeout for the tap-repeat feature (or
possibly a bug), which is something else entirely.  Not that it wouldn't be
nice to have Chrysalis able to configure all Qukeys' settings, of course.

mamapitufo, 2022-03-23:
> This sounds like an issue with the timeout for the tap-repeat feature (or possibly a bug), which is something else entirely. Not that it wouldn't be nice to have Chrysalis able to configure all Qukeys' settings, of course.

I removed my original comment because it turns out it was a bug, fixed on the latest firmware. It would still be nice to get this option, though.