keyboardio/Chrysalis #341: Default layer is not persisted after keyboard replug

szymonm, 2019-03-05:
**Describe the bug**
When I select some other layer than 0 for the default, then I replug the keyboard, the setting is gone.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to keyboard settings and select layer 3 as default.
2. Unplug the keyboard and connect it again.

**Expected behavior**
I expect the layer 3 will be the default again.

**Desktop (please complete the following information):**
 - OS: MacOs
 - Chrysalis Version: 0.4.0+77

algernon, 2019-03-05:
There should be a "Save changes" button on the same page:

[image]

Without pressing the button, the changed default is not saved to the keyboard. Is the problem reproducible with the button pressed prior to unplugging the keyboard?

szymonm, 2019-03-05:
Yes, it is the same even if I press the save button.

szymonm, 2019-03-05:
Also after pressing "save changes" after few seconds the "Default layer" value is back to "No default".

TreTuna, 2019-03-05:
This is probably the same bug I have been experiencing where it's not actually writing the default layer properly. @algernon, we talked a little about this in Discord and I was meaning to file this bug myself, but then my daughter was born and I forgot about everything, lol.

algernon, 2019-03-06:
Reproduced, kinda. I can set the default layer to "layer 0" or "layer 1", but setting it to "layer 2" blows things up. There's nothing useful on the dev console.

algernon, 2019-03-08:
This is likely a firmware issue, because setting layer 0, 1, and no default all work. Setting layer 2 and above seem to revert to no default.

szymonm, 2019-03-08:
Weird, because it worked before, when the default layer was set in the layout tab. I used layer three and didn't have this issue. I'm using keyboardio. cc: @obra

algernon, 2019-03-08:
Yeah, we (I) changed the firmware since, and that broke it. I have a fix incoming in a few minutes.

algernon, 2019-03-08:
keyboardio/Kaleidoscope#596 addresses this, will close when I'm updating the shipped firmware.

szymonm, 2019-03-08:
Thanks for the explanation! and for fixing of course!

commit 4fa8593c (2019-03-09): Update all the shipped firmware

Built from Chrysalis-Firmware-Bundle 0.5.0 (059a06a), using Kaleidoscope
master (622fa9c6).

Fixes #341 and fixes #346.
closed 2019-03-09 by algernon