keyboardio/Chrysalis #349: OneShot + ActiveModColor not showing for layers > 0

qayshp, 2019-03-11:
**Describe the bug**

**To Reproduce**
Steps to reproduce the behavior:
1. Set OSL on FN keys
2. Set OSM on Mod keys
3. Save to EEPROM
4. Activate a modifier key and notice that OSM and ActiveModColor work fine
5. Activate an FN and notice that active modifiers are no longer colored while in that layer (instead, keys like bksp, space, numbers, and hjkl are red.

**Expected behavior**
ActiveModColor should always work 🤷‍♂️. If it can't I would love to know why and what I need to disable to get it happy again perhaps with custom firmware again.

**Screenshots**
If applicable, add screenshots to help explain your problem. If possible, please also include a log from the "developer console". You can access it by pressing `Control + Shift + I`. When it appears, right click inside it, and select "Save as", and attach the saved file to this report.

**Desktop (please complete the following information):**
 - OS: macOS
 - Chrysalis Version: 0.5.0

**Additional context**
Firmware: experimental chrysalis 0.5.0 (also present in 0.4.0)
Keymap JSON in gist [here](https://gist.github.com/qayshp/1b4ec47c1cd66ed06aab7b2b865192a5).

algernon, 2019-03-15:
Does this happen for Layer#1 only? In the experimental & default firmwares, Layer#1 is the numpad layer, controlled by the NumPad plugin, which will do its own layer highlighting, overriding ActiveModColor.

Not sure how to best fix this... probably dropping NumPad from the experimental firmware would be the way to go.

algernon, 2019-03-16:
So, this is a combination of the NumPad plugin, and `ActiveModColor` being broken at the same time. For the former, disabling the plugin is probably the way to go, for the second, keyboardio/Kaleidoscope#609 is the fix.

I'll mark this issue resolved once the firmware files are updated.

qayshp, 2019-03-20:
By using a different layer number, as from #351, I am able to work around this issue now! Thanks for the information!

closed 2020-10-09 by algernon