keyboardio/Chrysalis #350: Unable to copy/paste from import/export textarea

qayshp, 2019-03-11:
**Describe the bug**
Unable to copy or paste from the textarea for import/export. I can edit the text manually, and copy from the DOM when I use the dev tools.

**To Reproduce**
Steps to reproduce the behavior:
1. Click to export a layer
2. Try to do anything with the text area.

**Expected behavior**
Copy, paste, select all, and cut should all work.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS
 - Chrysalis Version: 0.5.0

**Additional context**
Using release build downloaded from link on main readme. If I clone the repository and start Chrysalis using `yarn && yarn start` I am able to copy paste and get an edit menu.

algernon, 2019-03-15:
So, if I understand correctly, key bindings like `Ctrl+A`, `Ctrl+C`, etc (or their OSX equivalent) do not work when inside the text area?

Works fine under Linux, but I'll test under OSX later today. There's no code in Chrysalis that'd disable the key bindings, though.

There's no right-click context menu however, that is, indeed missing everywhere.

algernon, 2019-03-16:
Caleb on discord tested this further, `Ctrl+A`, `Ctrl+C`, etc do not work either.

qayshp, 2019-03-19:
> key bindings like Ctrl+A, Ctrl+C, etc (or their OSX equivalent) do not work when inside the text area?

correct, and the macOS equivalent is `cmd`+letter.

> `Ctrl+A`, `Ctrl+C`, etc do not work either.

I assume caleb was on windows. Probably unrelated, but curious: they sort of do work on my mac... when I place my cursor inside the text area, ctrl+a does jump to the start of the line, and ctrl+e to the end. I have no idea what does that though.

TreTuna, 2019-03-19:
Tested this as well, doesn't work on Mac for me. I'll look into it. If anything we could put a *Copy to clipboard" button on there really easily. Should be able to get to this tonight.

TreTuna, 2019-03-19:
Note: This is actually working for me in Chrysalis dev mode right now.

I do have a local branch with a `Copy to clipboard` button made as well if we want to introduce this.

[image]

algernon, 2019-03-19:
We'd need a "paste from clipboard" button too, to be complete. I'm pondering whether we want to replace import/export with a pair of open/save dialogs instead (the native kind of dialogs)? I don't think anyone would edit the data in-place anyway...

obra, 2019-03-19:
I frequently edit data in place on KLE. if the json panel eventually moves to something that doesn’t cover the keyboard, I expect I’d do it a lot. It’s “edit a text file” instead of “point and click” - it can be a very powerful interaction mechanism.

algernon, 2019-03-19:
You're officially a crazy person. :)

I'll try to look into why it doesn't work on release builds.

TreTuna, 2019-03-19:
@algernon I would possiblg wait until 0.5.1 is released then double check then put in the effort if it is still an issue. It works on the builds I'm making, just not in the current release, which is why I wasn't able to dig very far into it last night.

TreTuna, 2019-03-20:
Added a PR for the copy/paste buttons https://github.com/keyboardio/Chrysalis/pull/359

pm0u, 2020-05-16:
linux 5.6 / i3wm, chrysalis 0.7.5 I am able to copy and paste

algernon, 2021-01-25:
With the editor rework in 0.8.0, we directly load from / save to files, there's no longer any need to copy & paste, so this problem is no longer applicable. (And seems to have been fixed in an earlier release anyway.)

closed 2021-01-25 by algernon