keyboardio/Chrysalis #351: FN keys with OSL send some kind of code to the computer

qayshp, 2019-03-11:
**Describe the bug**
Pressing the FN keys 

**To Reproduce**
Steps to reproduce the behavior:
1. Type some text
2. Select the text
3. Press a FN key to OneShot a layer
4. Notice the text disappear

**Expected behavior**
FN keys should not send anything to the computer.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: macOS
 - Chrysalis Version: 0.5.0

**Additional context**
Keymap in use: https://gist.github.com/qayshp/1b4ec47c1cd66ed06aab7b2b865192a5

algernon, 2019-03-15:
Oh, I see what the problem is. It's *not* `OneShot`, but the `Numpad` plugin and OSX! Layer 1 is the numpad layer, and the Numpad plugin "controls" that: when switching to that layer, it sends a `NumLock` keycode, which OSX helpfully interprets as `Clear`.

This will be fixed soon, once keyboardio/Kaleidoscope#575 is merged and I build new firmware hexes.

qayshp, 2019-03-20:
Awesome to know that! Confirmed that if I change the layer numbering and use 6 as my default and 7 as my fn layer everything works fine*.

* - for some definition of fine. As in, this specifically is no longer an issue! But there are other problems.

closed 2020-10-09 by algernon