keyboardio/Chrysalis #354: Ability to add more t

ChristianKleineidam, 2019-03-16:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

TreTuna, 2019-03-16:
Adding layers is something that must be done on the firmware side. This is simply not possible in Chrysalis, and I'm not sure if it will be possible (@algernon would know better).

If you need more than 5 layers you will currently have to create your own firmware to load.

algernon, 2019-03-16:
Like @TreTuna said, there's no way to add new layers from Chrysalis, and no current plans to make it able to do so, thus, if you want or need more, you'll have to compile custom firmware at the moment.

It is technically possible to enhance the firmware so it allows creating additional layers, but that'd have way too many corner cases and error scenarios, so we didn't do it.

closed 2019-03-29 by TreTuna