keyboardio/Chrysalis #361: Blank screen after copy layer and save

fluffy-critter, 2019-03-21:
**Describe the bug**

From commit 9cb6de187066279de9bc0fc584945619516bc9a4, got a blank screen after copying a layer and then switching layers

**To Reproduce**

Steps to reproduce the behavior:

1. Went to layer 1
2. Copied from layer 3
3. Saved to keyboard
4. Switched to layer 0

**Expected behavior**

Expected to see the layer 0 editor, got a white screen instead

**Desktop (please complete the following information):**
 - OS: macOS
 - Chrysalis Version: master @ 9cb6de187066279de9bc0fc584945619516bc9a4

Log attached. Also includes previous export of EEPROM contents using the old internal method, which might be helfpul for diagnosing I guess.

[localhost-1553156928510.log](https://github.com/keyboardio/Chrysalis/files/2991268/localhost-1553156928510.log)

ryan-williams, 2020-05-03:
I believe I'm seeing this as well:

[image]

<details>
<summary>Log</summary>
renderer.js:83 focus.find: portList: Array(7) devices: Array(9)
renderer.js:83 focus.find: found_devices: Array(1)
renderer.js:83 Connecting to /dev/tty.usbmodemCkbio01E1
renderer.js:83 Probing for Focus support...
renderer.js:83 focus.request: help
renderer.js:83 focus: incoming data: keymap.custom
keymap.default
keymap.onlyCustom
renderer.js:83 focus: incoming data: help
renderer.js:83 focus: incoming data: settings.defaultLayer
settings.valid?
settings.version
settings.crc
renderer.js:83 focus: incoming data: eeprom.contents
eeprom.free
renderer.js:83 focus: incoming data: palette
renderer.js:83 focus: incoming data: colormap.map
renderer.js:83 focus: incoming data: .
renderer.js:83 focus.request: settings.defaultLayer
renderer.js:83 focus: incoming data: .
renderer.js:83 focus: incoming data: 0 
renderer.js:83 focus: incoming data: .
renderer.js:83 focus: incoming data: .
renderer.js:83 focus.request: keymap.default
renderer.js:83 focus: incoming data: 65535 30 31 32 33 34 17152 224 228 24577 35 36 37 38 39 17409 53 20 26 8 21 23 43 42 44 40 28 24 12 18 19 46 75 4 22 7 9 10 41 227 226 230 11 13 14 15 51 52 78 29 27 6 25 5 17452 225 229 17452 17 16 54 55 56 45 65535 65535 65535 65535 65535 65535 65535 65535 65535 24576 65535 36 37 38 86 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 33 34 35 87 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 30 31 32 46 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 39 55 85 84 40 65535 58 59 60 61 62 57 65535 65535 22710 63 64 65 66 67 68 43 65535 20481 65535 20546 20576 20521 76 40 22733 22709 2095 2096 47 48 69 74 20484 20482 20488 20545 20517 20522 65535 65535 101 80 81 82 79 65535 65535 77 70 73 65535 20548 20518 65535 65535 65535 65535 19682 23786 23785 65535 49 2097 
renderer.js:83 focus: incoming data: .
renderer.js:83 focus: incoming data: .
renderer.js:83 focus.request: keymap.custom
renderer.js:83 focus: incoming data: 65535 30 31 32 33 34 17152 224 228 24577 35 36 37 38 39 17409 53 20 26 8 21 23 76 40 44 40 28 24 12 18 19 46 43 4 22 7 9 10 42 227 226 230 11 13 14 15 51 52 41 29 27 6 25 5 17410 225 229 17452 17 16 54 55 56 45 65535 65535 65535 65535 65535 65535 65535 65535 65535 24576 65535 36 37 38 86 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 33 34 35 87 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 30 31 32 46 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 39 55 85 84 40 65535 58 59 60 61 62 57 65535 65535 22710 63 64 65 66 67 68 78 75 20481 65535 20546 20576 20521 40 40 22733 22709 2095 2096 47 48 69 74 20484 20482 20488 20545 20517 20522 65535 65535 101 80 81 82 79 65535 65535 77 70 73 65535 20548 20518 65535 65535 65535 65535 19682 23786 23785 65535 49 2097 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 
renderer.js:83 focus: incoming data: .
renderer.js:83 focus.request: keymap.onlyCustom
renderer.js:83 focus: incoming data: .
renderer.js:83 focus: incoming data: 1 
renderer.js:83 focus: incoming data: .
renderer.js:83 focus.request: palette
renderer.js:83 focus: incoming data: .
renderer.js:83 focus: incoming data: 208 2 27 245 166 35 184 233 134 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
renderer.js:83 focus: incoming data: .
renderer.js:83 focus.request: colormap.map
renderer.js:83 focus: incoming data: .
renderer.js:83 focus: incoming data: 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 2 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 2 1 15 0 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 15 
renderer.js:83 focus: incoming data: .
renderer.js:83 focus: incoming data: .
renderer.js:83 focus.request: keymap.onlyCustom 1
renderer.js:83 focus: incoming data: .
[... cut ...]
    at Di (webpack:/node_modules/react-dom/cjs/react-dom.production.min.js:249)
    at Yh (webpack:/node_modules/react-dom/cjs/react-dom.production.min.js:248)
    at Xh (webpack:/node_modules/react-dom/cjs/react-dom.production.min.js:245)
    at Hl (webpack:/node_modules/react-dom/cjs/react-dom.production.min.js:243)

</details>

It seems to be happening mostly after I (successfully) save to my keyboard; I think it's been happening in other situations too. The only workaround I've found is quitting and restarting Chrysalis. Any suggestions are welcome. Thanks!

algernon, 2022-05-17:
I believe this was fixed a while ago, but in any case, we no longer support per-layer import or copying.

closed 2022-05-17 by algernon