keyboardio/Chrysalis #362: provide a dvorak starting layer

bjornicus, 2019-03-21:
As noted in #268 having a way to share layers would be great, but in the shorter term just having a starting json used for import/export) for the common alternate layouts (e.g the same layouts as https://github.com/keyboardio/Kaleidoscope/wiki/Common-Alternate-Layouts) would be really useful.  I can probably do the dvorak one and submit a  PR with the layer 0 if you give some guidance on a good place to add it.  Perhaps a top level "sample_layouts" folder?

bjornicus, 2019-03-21:
hmm, except I can't seem to copy the json it's showing me when I press import/export....

Penthious, 2019-03-26:
@bjornicus This is what I use for dvorak. I would also be willing to set up something where at the click of a button it changes all the keys to dvorak or any other layout. Would just need some help with where I should start.

```
{
  "keymap": [
    {
      "keyCode": 65535,
      "label": "",
      "verbose": "Transparent"
    },
    {
      "keyCode": 30,
      "label": "1"
    },
    {
      "keyCode": 31,
      "label": "2"
    },
    {
      "keyCode": 32,
      "label": "3"
    },
    {
      "keyCode": 33,
      "label": "4"
    },
    {
      "keyCode": 34,
      "label": "5"
    },
    {
      "keyCode": 17152,
      "label": "Next",
      "extraLabel": "LEDEff."
    },
    {
      "keyCode": 224,
      "label": "LCtrl",
      "verbose": "Left Control"
    },
    {
      "keyCode": 228,
      "label": "RCtrl",
      "verbose": "Right Control"
    },
    {
      "keyCode": 24577,
      "label": "#1",
      "extraLabel": "Macro"
    },
    {
      "keyCode": 35,
      "label": "6"
    },
    {
      "keyCode": 36,
      "label": "7"
    },
    {
      "keyCode": 37,
      "label": "8"
    },
    {
      "keyCode": 38,
      "label": "9"
    },
    {
      "keyCode": 39,
      "label": "0"
    },
[... 518 lines cut ...]
      15,
      15,
      15,
      15,
      15,
      15,
      15,
      15,
      15,
      15,
      15,
      15,
      15,
      15,
      15,
      15,
      15,
      15,
      15,
      15
    ]
  ],
  "palette": []
}
```

TreTuna, 2019-03-26:
I made a quick PR that adds buttons for all the default layouts. https://github.com/keyboardio/Chrysalis/pull/366

TreTuna, 2019-03-30:
Solved in #366

closed 2019-03-30 by TreTuna