keyboardio/Chrysalis #363: Blank screen after pasting layout and switching layer

fluffy-critter, 2019-03-21:
**Describe the bug**

Get a blank window after pasting a JSON layout from the clipboard

**To Reproduce**

Steps to reproduce the behavior:

1. go to layer 0, select copy/paste 
2. paste from clipboard
3. press 'okay'
4. go to layer 1

**Expected behavior**

Expected to be able to then edit layer 1

**Screenshots**

[localhost-1553201025958.log](https://github.com/keyboardio/Chrysalis/files/2993912/localhost-1553201025958.log)

**Desktop (please complete the following information):**
 - OS: macOS
 - Chrysalis Version: 9cb6de1

**Additional context**
Add any other context about the problem here.

mattvenn, 2019-05-14:
confirmed. Also I see this with copying to clipboard as well.

obra, 2019-07-17:
This should not have been resolved if the patch was applied to Dygma's fork, but not the core.

0957758592, 2019-08-04:
It looks like this issue should be closed, cause were being merged.

fluffy-critter, 2019-08-04:
As a note, if you put text like "Fixes #363" into the PR comment, then the issue will automatically close when the merge happens.

0957758592, 2019-08-04:
@fluffy-critter 
Thank you!

closed 2019-08-19 by obra