keyboardio/Chrysalis #365: I can't access "Firmware Update" and "Keyboard Settings" screens

Praedes, 2019-03-22:
Hello,

I'm on Ubuntu 18.04.2 LTS. 
Chrysalis 0.5.0+92.

My Model01 is recognized (totally new, I never install anything on it).
On Chrysalis, I click on "Connect", the "Firmware update" appear. But when I click on "Firmware Update" or "Keyboard Settings", nothing happen.
I'm stuck here on the main screen : 
[image]

The permissions on /dev/ttyACM0 seems good : 
[image]

I restart Chrysalis, restart the computer, unplug replug the keyboard...

An interesting detail maybe, when I click on "Firmware Update", I stay on the main screen but the "Firmware Update" is highlight on the menu : 
[image]

Thank you,
Cordialy

caleb2017x2, 2019-03-23:
I'm having the same problem on a Mac running macOS 10.14.3 and the latest version of Chrysalis.

CADTraveller, 2019-04-09:
Me too, Win10 Version 1903

tsh-xx, 2019-04-17:
I managed to turn on dev tools (after running as administrator), these are the errors I see in the console:
`Uncaught TypeError: Cannot read property 'substring' of undefined
    at ClientRequest.substring (webpack:/src/renderer/utils/getLatestVersion.js:43)
    at Object.onceWrapper (events.js:273)
    at ClientRequest.emit (events.js:182)
    at HTTPParser.parserOnIncomingClient (_http_client.js:555)
    at HTTPParser.parserOnHeadersComplete (_http_common.js:109)
    at Socket.socketOnData (_http_client.js:441)
    at Socket.emit (events.js:182)
    at addChunk (_stream_readable.js:283)
    at readableAddChunk (_stream_readable.js:264)
    at Socket.Readable.push (_stream_readable.js:219)
App.js:144 Connecting to COM3
App.js:149 Probing for Focus support...
30getLatestVersion.js:43 Uncaught TypeError: Cannot read property 'substring' of undefined
    at ClientRequest.substring (webpack:/src/renderer/utils/getLatestVersion.js:43)
    at Object.onceWrapper (events.js:273)
    at ClientRequest.emit (events.js:182)
    at HTTPParser.parserOnIncomingClient [as onIncoming] (_http_client.js:555)
    at HTTPParser.parserOnHeadersComplete (_http_common.js:109)
    at Socket.socketOnData (_http_client.js:441)
    at Socket.emit (events.js:182)
    at addChunk (_stream_readable.js:283)
    at readableAddChunk (_stream_readable.js:264)
    at Socket.Readable.push (_stream_readable.js:219)
Ca.a.request.t @ getLatestVersion.js:43
onceWrapper @ events.js:273
emit @ events.js:182
parserOnIncomingClient @ _http_client.js:555
parserOnHeadersComplete @ _http_common.js:109
socketOnData @ _http_client.js:441
emit @ events.js:182
addChunk @ _stream_readable.js:283
readableAddChunk @ _stream_readable.js:264
Readable.push @ _stream_readable.js:219
onStreamRead @ internal/stream_base_commons.js:94`

I also got `Uncaught (in promise) Error: Opening COM3: Access denied` when I closed and re-started.

This is win10, 1809, and an early model with original firmware.

algernon, 2019-04-17:
Oh, looks like there are two problems here:

- There is no error handling in the version check
- Chrysalis doesn't have access to the serial port

The latter you can work around by running Chrysalis as administrator.

leavesofgrass, 2019-04-17:
Perhaps this is my issue with the 0.5 AppImage in Ubuntu. I have uninstalled modem-manager but my model01 hangs while connecting. I can't access layouts or update the firmware in Linux but everything works swimmingly in Windows.

mbutz, 2019-05-01:
Hi, I am experiencing the same issues with my freshly delivered Model 01 and the first use of Chrysalis. I can't actually use/open any of the menu options (). Thought the keyboard is recognized. There is no difference if I start Chrysalis as admin. I also don't see how I can activate the developer mode to get some error reporting.

[image]

Nevertheless the syslog says:

```
May  1 12:18:18 mbox ModemManager[1052]: <info>  Creating modem with plugin 'Generic' and '1' ports
May  1 12:18:18 mbox ModemManager[1052]: <warn>  Could not grab port (tty/ttyACM0): 'Cannot add port 'tty/ttyACM0', unhandled serial type'
May  1 12:18:18 mbox ModemManager[1052]: <warn>  Couldn't create modem for device at '/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.3': Failed to find primary AT port
```

I am under Linux Mint 18.1 (which is Ubuntu 16.4); I hope, this is not an issue because right now I wouldn't want to update.

Cheers
Martin

leavesofgrass, 2019-05-01:
@mbutz it looks like the modem manager service is running that will prevent the keyboard from connecting. 
Try uninstalling modem manager or disabling the service?

mbutz, 2019-05-01:
Hi @leavesofgrass,

thanks for the hint. I did uninstall the modem manager, reboot but no success. I don't have those syslog messages anymore but I still can't access any of the menu item (both as normal user and root).

mariusmorabosch, 2019-05-03:
Hi! Having the same problem here on windows 10. Running the software as administrator does not fix the issue. It's a brand new model 01.

mbutz, 2019-05-04:
@caleb2017x2: Open the terminal and
```
sudo bash # provide password
cd /Applications/Chrysalis.app/Contents/MacOS # go to program directory
./Chrysalis # start program
```
I did that on my Mac running OS X El Capitan (10.11.6). Unfortunately no difference to the above described behaviour under Linux.

blkbsstt, 2019-05-04:
I'm having the same problem with 0.5.0 on a new Model 01, OSX 10.14.3. I also couldn't get it to work on an Ubuntu box. The console doesn't show any errors.

leavesofgrass, 2019-05-04:
In my further testing, my model 01 only connects in Windows 10 not MacOS or
Ubuntu. Building from source may work but I haven't had time to test that.
Hopefully the next release will work better in Linux, working on firmware
requires too much rebooting right now.

Ptepp1c, 2019-05-06:
As mentioned in discord having the same issue with windows 10 with new Model 01, I have also tried to use Chrysalis 0.4 but that didn't work either (though with a different sort of error.) I tried running as Admin in both versions.

Effectively Chrysalis does not work whatsoever for me currently as it just seems stuck before it loads the layers.

mariusmorabosch, 2019-05-09:
@algernon @TreTuna Can we get some feedback on this? The software is basically useless for me and other people. I mention you as I see you are the most recent active devs on here :)

TreTuna, 2019-05-09:
@algernon is the person with the most knowledge on this across platforms. I have only used it on the latest version of MacOS and have not been able to reproduce this issue there.

obra, 2019-05-09:
(And @algernon is in the process of moving house, so he's been less
available than usual)
ᐧ

TreTuna, 2019-05-09:
(And I have a new baby and new job so it's been little time to help with OSS right now, sadly)

patchmonkey, 2019-05-10:
I'm having this issue with Chrysalis 0.5.0 on Windows 10. I was able to install, flash, and configure on my OS X laptop earlier, but when connecting it back to Windows 10, I experience the same issues as described above.

obra, 2019-05-10:
For folks running into this with 0.5.0 on Windows, can you try the latest
nightly build?

http://kaleidoscope-builds.s3-website-us-west-2.amazonaws.com/Chrysalis/latest/Chrysalis.exe

mariusmorabosch, 2019-05-10:
Same issue persists for me with the nightly build :(

algernon, 2019-05-10:
Apologies for being absent lately, I'm on it now. From what I can tell, the main issue is with the version check, I'll have a fix for that in a bit.

algernon, 2019-05-10:
Interestingly, the version check does *not* appear to be the issue. While that ends up being printed as an error, things continue working regardless, as far as I can tell.

On the forums, someone said that 0.5.0 (without the +94) worked, but the latest build did not. So perhaps there's something inbetween that broke Chrysalis, but I can't reproduce right now. I'll push a fix for the version check though, just in case.

tsh-xx, 2019-05-10:
@algernon Yes, confirmed the nightly build does not work, but 0.5.0 release version does at least get to the stage of telling me it needs to flash new firmware (which I'm just bout to try). This is on Win10, and I don't need to run as admin to get this result.

mbutz, 2019-05-10:
Hi,

menu options and functions seem to be available with [Chrysalis 0.5.0](https://github.com/keyboardio/Chrysalis/releases/tag/chrysalis-0.5.0) as noted [here](https://community.keyboard.io/t/keyboardio-without-programming-knowledge/2817/12).

leavesofgrass, 2019-05-10:
The 0.5.0 AppImage is also working for me in Ubuntu while version 0.5.0+92.would not connect to my model01.

mariusmorabosch, 2019-05-11:
Hi, can confirm that 0.5.0 works for me too (without the +94) now! So definetely something on that build went wrong!

mattvenn, 2019-05-13:
latest snapshot from master not working (hangs after clicking connect - debug console says 'probing for focus'), but release 0.5 works fine.

we built on windows 10 by:

installing node 10.15.3
yarn 1.16.0
npm install -g windows-build-tools
yarn
yarn build:win

this also fails. But yarn start works.

obra, 2019-05-13:
Ok. The next step is for somebody to "bisect" to figure out which revision
broke things.

Does
https://kaleidoscope-builds.s3.us-west-2.amazonaws.com/Chrysalis/90/Chrysalis-0.5.0%2B90.exe
work
correctly?

If it doesn't, how about
https://kaleidoscope-builds.s3.us-west-2.amazonaws.com/Chrysalis/88/Chrysalis-0.5.0%2B88.exe

If it does, how about
https://kaleidoscope-builds.s3.us-west-2.amazonaws.com/Chrysalis/93/Chrysalis-0.5.0%2B93.exe
?
ᐧ

Ptepp1c, 2019-05-13:
+96 does not work 
(build from main page)
+93 does not work 
(3rd link)
+90 does work correctly (initial loading is fine at least)
(1st link)

obra, 2019-05-13:
ok! thank you! can you try 91 and 92? There are two places you need to change the link.

Ptepp1c, 2019-05-13:
Neither work correctly, in case it helps +91 and +92 are both 200mb when installed.

+0.90 which does work is 198mb

obra, 2019-05-13:
Awesome. Thank you! Now we know exactly which change broke things

algernon, 2019-05-17:
So it looks like b8fa55db3b9e174d68b1b0e350ff618be8631951 (the commit +91 is built from) is the culprit. I don't really see how the code changes would break anything, so perhaps its one of the dependencies.

algernon, 2019-05-17:
Also, @mattvenn also found that running `yarn run start` still works, even on master. To me, this suggests that we could try backing out the `webpack` update, or perhaps `babel`, but my gut feeling says its webpack.

algernon, 2019-05-17:
Oh, and I can reproduce the same issue on Linux with the AppImage, so it's not even windows specific. Neat!

algernon, 2019-05-17:
Reverting `package.json` and `yarn.lock` to the commit before b8fa55db3b9e174d68b1b0e350ff618be8631951, and adding back `electron-window-state` does fix the issue. Now to figure out which one of the updates broke the build (so far it seems its not webpack, nor electron, but we'll see).

algernon, 2019-05-17:
Narrowed it down to `electron-webpack`, `electron-builder` and `webpack`. A combination of upgrading those broke our prod builds.

algernon, 2019-05-17:
Narrowed it down to `electron-webpack`, but only kinda. If I take the `package.json` and `yarn.lock` files from the commit prior to breakage, and back out everything from `package.json` but `electron-webpack`, things work. But as soon as I regenerate `yarn.lock` from scratch, we break again.

So the problematic part is *not* `electron-webpack`, but some other transient dependency. That's going to be a pain to figure out, as the `yarn.lock` diff between working and non-working is huge.

algernon, 2019-05-17:
Pushed a workaround that fixes things for me, it should be built by Travis in an hour or two. I don't have the brains to debug which transient dependency broke us, so we'll have to figure that out sometime later. But the `package.json` / `yarn.lock` pair I pushed just now made both the AppImage and the Windows exe work for me.

I feel it's more important to have a working production build than solving the problem correctly, so we have a temporary workaround for the time being.

tsh-xx, 2019-05-18:
Doesn't seem to work (nightly build +98). Different to before, but I don't seem to be able to get past 'must update firmware'. This _is_ running as administrator.
```
internal/process/next_tick.js:68 Uncaught (in promise) Error: Opening COM3: Access denied
D:\tsh\AppData\Local…rysalis-flash.js:72 baud update
D:\tsh\AppData\Local…rysalis-flash.js:77 dtr on
D:\tsh\AppData\Local…rysalis-flash.js:82 dtr off
D:\tsh\AppData\Local…ib\connection.js:23 found Keyboardio Model 01 on port COM3
D:\tsh\AppData\Local…no\lib\avr109.js:42 reset complete.
D:\tsh\AppData\Local…rysalis-flash.js:45 Error: could not open board on COM3
    at D:\tsh\AppData\Local…b\connection.js:152
    at next (D:\tsh\AppData\Local…es\awty\index.js:39)
    at SerialPort.<anonymous> (D:\tsh\AppData\Local…b\connection.js:143)
    at SerialPort._error (D:\tsh\AppData\Local…b\serialport.js:199)
    at binding.open.then (D:\tsh\AppData\Local…b\serialport.js:239)
FirmwareUpdate.js:150 Error: could not open board on COM3
    at D:\tsh\AppData\Local…b\connection.js:152
    at next (D:\tsh\AppData\Local…es\awty\index.js:39)
    at SerialPort.<anonymous> (D:\tsh\AppData\Local…b\connection.js:143)
    at SerialPort._error (D:\tsh\AppData\Local…b\serialport.js:199)
    at binding.open.then (D:\tsh\AppData\Local…b\serialport.js:239)
events.js:167 Uncaught Error: Port is not open
    at SerialPort.close (D:\tsh\AppData\Local…b\serialport.js:413)
    at avrgirl.flash (D:\tsh\AppData\Local…rysalis-flash.js:48)
    at D:\tsh\AppData\Local…no\lib\avr109.js:45
    at D:\tsh\AppData\Local…b\connection.js:155
    at next (D:\tsh\AppData\Local…es\awty\index.js:39)
    at SerialPort.<anonymous> (D:\tsh\AppData\Local…b\connection.js:143)
    at SerialPort._error (D:\tsh\AppData\Local…b\serialport.js:199)
    at binding.open.then (D:\tsh\AppData\Local…b\serialport.js:239)
events.js:167 Uncaught Error: Port is not open
    at SerialPort.close (D:\tsh\AppData\Local…b\serialport.js:413)
    at Focus.close (D:\tsh\AppData\Local…ysalis-focus.js:112)
    at Object.close [as onDisconnect] (webpack:/src/renderer/App.js:177)
    at Object.onDisconnect [as onClick] (webpack:/src/rendere…rmwareUpdate.js:155)
﻿
```

​

algernon, 2020-04-29:
So the original issue should be fixed. The flashing issue is going to be fixed in #508: what happens there is that we reset the keyboard, and it comes back on a different port number, yet, we try to connect to the old one. As part of #508, we will re-scan devices and detect the new port number. That will fix the flashing issue.

Once #508 is merged, I'll be closing this one too.

algernon, 2020-10-09:
As #508 has been merged a while ago, I'm closing this too.

closed 2020-10-09 by algernon