keyboardio/Chrysalis #367: Better error handling/reporting

TreTuna, 2019-03-26:
**Is your feature request related to a problem? Please describe.**
Currently, we get a lot of issues coming in where the user has landed on a blank screen because an error has occurred and React unmounts the entire application. After which we continually ask the user to open dev tools and copy/paste or screenshot what's going on.

**Describe the solution you'd like**
1) Add error boundaries in the application with a fallback UI component that gives the user instructions of how to report the problem, a link to Github, and a button that will copy the error to their clipboard for easy reporting.

2) Integrate [Sentry-electron](https://github.com/getsentry/sentry-electron) for full-featured error reporting/logging. Sentry is a great tool for automatic error reporting/logging, we can use a free developer account for now and submit for a [free open source license](https://sentry.io/for/open-source/).

TreTuna, 2019-03-29:
@algernon and @obra what do you think about adding [Sentry](https://sentry.io/welcome/) for error reporting/logging? I believe I can get this all set up for us now that I am a part of the Keyboardio org/Chrysalis team. I want to work get going on this issue next and get us some better error handling.

algernon, 2019-03-29:
I'm a bit vary of automatic error reporting. Not because its not useful (it is), but because it's too easy to leak sensitive information (like, reporting eeprom contents has a fair chance of being sensitive in the not too distant future), and it all happens without user consent. I know we won't be doing anything evil with the data, but... If we do this, I'd make it opt-in. But I'd rather stick to asking our users to report errors explicitly, even if that makes debugging harder. I'd opt for solution 1) instead, I'm afraid.

TreTuna, 2019-03-29:
I understand completely and all good there!

algernon, 2021-01-25:
We now have a "System Information" screen, which can create a bundle for us, with system info, logs, and whatnot. I think this feature adequately addresses this feature request.

closed 2021-01-25 by algernon