keyboardio/Chrysalis #368: Default window size is small

TreTuna, 2019-03-29:
Currently we use the Electron default window size of 800x600 when we boot up. On HD+ sized screens (1920 x 1080 and up) it is difficult to read the font on the keys. I've found a good default for myself is to step up to 1200x900.

algernon, 2019-03-29:
Increasing the default size to 1200x900 is a good idea, and a simple change to make, too. In the longer run, we should probably save the window position, size and state (maximised or not), and restore that on startup. The [electron-window-state](https://www.npmjs.com/package/electron-window-state) module might be useful for this purpose.

TreTuna, 2019-03-29:
Hm... I'll have to play around with that package.

TreTuna, 2019-03-30:
Solved in #370

closed 2019-03-30 by TreTuna