keyboardio/Chrysalis #372: Show proper characters in the select boxes when shift is checked

TreTuna, 2019-04-12:
**Is your feature request related to a problem? Please describe.**
Currently, when you set a key to either a number or punctuation keycode, then click on the checkbox to hold shift, the key itself shows the proper character, but the key select box still shows the non-shifted characters. ie: With shift-clicked, the char select shows a semicolon, though the key itself shows a colon, which is the character that is output by shift+semicolon.

**Describe the solution you'd like**
The solution I can think of at the moment would be to change the characters in the key select boxes to show the proper character depending on if the shift checkbox is checked or now. ie: if the shift checkbox is marked true, the key selector for a semicolon would turn into a colon.

**Describe alternatives you've considered**
none at this time

**Additional context**

[image]

[image]

obra, 2019-04-12:
This makes sense, though we'll need to figure out something clever when
doing internationalization.
ᐧ

algernon, 2022-05-17:
We have redesigned the keymap editor since this issue was reported, and the problem does not exist in the new version:

[image]

closed 2022-05-17 by algernon