keyboardio/Chrysalis #375: Document how to use Chrysalis

algernon, 2019-04-23:
We should have documentation about how to use Chrysalis, how to reassign keys, use the colormap, and so on. This documentation should also include a troubleshooting chapter for common issues (permission issues, ModemManager, etc).

Ptepp1c, 2019-05-15:
I am happy to help with this, while I am not a programmer this may actually be a positive for documentation :) .

TreTuna, 2019-05-16:
That would be great. I am happy to help with any questions you have while doing this.

algernon, 2019-05-17:
Same here, I'm very happy to answer any questions you may have. Do we want this as a WiKi page, a blog post, or something else?

In any case, the app itself should have an in-app link to the documentation.

Ptepp1c, 2019-05-18:
I think the main things from my point of view will be rough timelines for major changes to how the software works, for example I know that there is a lot of thought going into designing a new way for handling colour changes, if that is happening within a month, there is little point going into detail documenting how the current version works, if its six months down the line then it makes sense to write up a guide for the current version. 

As for format a wiki may be a good way to approach as  corrections can be made by multiple people easily (I presume anyway, have written wikis). 

I think the major thing I am unsure about is version control, for uses coming in afresh obviously they can go through everything, but for current users it will be finding a very good way to detail changes and point to new guidance without having to wade through everything that stayed the same.

I will probably be typing in word, proofreading then putting it up somewhere for people to proof read, suggest changes etc.

jfmcdowell, 2019-05-22:
I previously offered to help as well. Over the rest of the week I will be getting a shell Wiki together. Please feel free to modify.

aspiers, 2020-12-31:
One of the things which needs to be documented is which Kaleidoscope plugins are supported.z